ttn-zh/ic880a-gateway #5: Improvement on antenna

yerpj, 2016-04-05:
Hello guys.
First I would like to say that your job is impressive! I wish I lived in Zurich. We do not have most coverage here in Lausanne and Yverdon, even if I heard that [Octanis](http://octanis.org/) installed a gateway recently near the EPFL.

I am not a RF guy, but I work with some of them.In my opinion it should be of importance to specify in your wiki that the antenna SHOULD be a 1/2wavelength. If 1/4wavelength, the mounting instructions are missing a rock-solid copper-foil ground plane.

Once more, thumbs up for all the work you are doing!

gonzalocasas, 2016-04-10:
Hi @yerpj! Thanks for the feedback! I added the 1/2 wavelength remark. Is it correct as it is now? I also added a few links to other higher gain antennas from Aliexpress.

closed 2016-04-11 by yerpj

yerpj, 2016-04-11:
Yes that's better! This will avoid confusion. Thank you.