ttn-zh/ic880a-gateway #8: https://github.com/ttn-zh/ic880a-gateway.git/ not found

skramer, 2016-07-09:
git clone https://github.com/ttn-zh/ic880a-gateway.git ~/ic880a-gateway
after entering github user/pass

and I get back a message:
fatal: repository git clone https://github.com/ttn-zh/ic880a-gateway.git/ not found

The above command comes from:
https://github.com/ttn-zh/ic880a-gateway
section: Clone the installer and start the installation

Can anyone advise what the issue is.. 
Thank you in advance

gonzalocasas, 2016-07-09:
Hi @skramer, this doesn't look like a problem with the repo, something seems weird on your git setup. I just ran the same command and it worked without issues.

Did you follow the instructions on [the wiki](https://github.com/ttn-zh/ic880a-gateway/wiki)? Are you starting from a clean Raspbian Jessie Lite?

Normally, git should not ask for your user and pass when you clone this repo, so that is already a warning sign that your git setup is doing unexpected stuff. If you can't find the root cause of the issue, I'd suggest restarting the process, get a new clean image of raspbian, and follow the instructions on the wiki.

Let us know how it goes.

skramer, 2016-07-09:
Thank you kindly for responding... 
The answer is YES we are following the wiki (we would say) exactly.. We have had zero errors or changes to the install flow... starting with a new Rasperian image and following along.. 
However since you are clear about this bad behavior we will start again ... and note our process.. 
BTW .. is ttn-zh part of 'The Things Network' .. we are looking forward towards participating.. (as soon as we get over this error.. smiles..............)Thank you Steven

gonzalocasas, 2016-07-09:
Hi Steven! `ttn-zh` is the Zurich community of The Things Network. It's great that you want to join in! Maybe you'd be interested in joining the global Slack: http://slack.thethingsnetwork.org/ and -in case you don't have one already- create a community page and community groups for your area.

Let me know if this time it worked, if it doesn't, we can keep checking, but I have executed these scripts a million times, and I know at least one other guy that ran them today successfully, so I am not sure what could be different in your case. The only thing that comes to mind as one possibility is that you're behind a very aggressive firewall that is messing up with the SSL connection, but is a really long shot.

gonzalocasas, 2016-07-09:
One thing to check is whether the ssh stuff in proper. Run the following command:

    $ ssh -T git@github.com

You should receive a message like this:

    Hi skramer! You've successfully authenticated, but GitHub does not provide shell access.

skramer, 2016-07-09:
Gonzalo,Thank you for your attention ... 
a) By a careful reinstall (starting from the top we. have gotten past this error..
and I think I have a working Gateway.. However having never worked with LoRa before >... 
I am not certain how to 'Configure' or 'Test' the gateway... 
Can you help me with this please.. 
Also Ill look into slack... Our business is thinking of sponsoring 'Coral Springs Fl' .. I think only 3 gateways will cover the city and would enable us to offer services not currently available.. (as you know)
Before we can do that we want to completely understand the technology to that end we have a couple of LoRa nodes to test to this gateway. 
Thank you. your assistance is welcome.. smiles............Thank you Steven

skramer, 2016-07-10:
I see our gateway as 'active' here:B827EBFFFE9D3258 active   active Fantastic.. now time to learn to use it.. Let me know what we should do next... smiles..............Thank you Steven

gonzalocasas, 2016-07-10:
Hi Steven! Great stuff! 
I see your gateway shows up as active in the map (http://ha-23.eradus.eu/croft.html)!
[image]

Now you should build a little test node and start playing around with it! Check the TTN wiki for more general info: http://staging.thethingsnetwork.org/wiki/Home 

I do have one question thou: could it be that you have the wrong concentrator board? If you are located in USA, the frequency assigned for LoRa in the US is 915Mhz, and as far as I know, the ic880a concentrator board is only manufactured for the 868Mhz frequency. This means, you probably can use it for testing, but it is not really correct to use it for real, because it won't work with nodes built for the US (and actually, I'm not even sure 868 is legal over there).

closed 2016-07-10 by gonzalocasas

skramer, 2016-07-10:
Gonzalo,
Thank you ... arrrrrrrrrrrrrrrrrrrrggggggggggggggggggggggg................... I did not realize the 868/915 thing... sigh.. So what are my options for a DIY 915 Gateway..???
BTW I do intend to purchase a 'The Things Network Gateways' though I understand not available now.. When might they?
Getting to work on setting up a node. We purchased Adafruit Feather 32u4 RFM95 LoRa Radio - 868 or 915 MHz and work to implement things4u/LoRa-LMIC-1.51 .... Your thoughts.. thank you Steven

  
|  
|   
|   
|   |    |

   |

  |
|  
|    |  
things4u/LoRa-LMIC-1.51
 LoRa-LMIC-1.51 - Port of LMIC-1.5 library tp Atmega 328 and ESP6822, including alternative AES encription library  |   |

  |

  |

gonzalocasas, 2016-07-10:
I am not totally sure, but I think you can build this one using a 915Mhz Lora card: https://github.com/mirakonta/lora_gateway/wiki
But actually, I'd suggest you use what you have for testing and pre-order one of the gateways from The Things Network: https://shop.thethingsnetwork.com/index.php/product/the-things-gateway/  We're not far from the estimated shipping date (around August).

Regarding nodes, if your soldering skills are good, then I'd suggest you get the Microchip RN2903 (or the RN2483 if you want the 868Mhz version to test), if not, the Adafruit that contains the RFM95 is probably good (but it's a bit more work to get going with the LMIC stack than it is with the microchip one). But if you do get this one, use the following port of LMiC: https://github.com/matthijskooijman/arduino-lmic

skramer, 2016-07-11:
Gonzalo,
Ok yes .. Ill await your 'The Things Network Gateway and get the 915 version... I HAD ordered and canceled my order when I did not want to wait till August.. and would love to support 'The Things Network'
Regarding node... I have tried the arduino-limic   .. giving me a fit... Difficulty in uploading and there seems to be some incompatibility .. so I'll have to work this and get some issues documented.. 
In the mean time .. I am interested in your comment on the RN2483.. yes have very good soldering skills .. make all kinds of connected soldered prototypes.. What link should I follow for instructions.. ? Let me know please 
Thank you Steven

gonzalocasas, 2016-07-11:
Here's the datasheet for the microchip module: http://www.microchip.com/wwwproducts/en/RN2483 You can order using microchipdirect or any of the many resellers: http://www.findchips.com/search/rn2483

Keep in mind that RN2483 is the 868Mhz version, which will work for you to test now with the iC880a, but eventually, you should get the RN2903 version of the same module, which works on 915Mhz for US.

skramer, 2016-07-11:
Ok just purchased two RN2483's and will follow this setup .. Seems rather straight away .. need to solder 5 pins..... 
I should have the parts by Thursday .. and we will see how it goes.. Thank you for your kind guidance... 
Steven Kramer

skramer, 2016-07-13:
Gonzalo,
I got my RN2483s today. Got them Wired up .. Installed the RN2483 librariesOpened the TXfastViaTTN.ino Added   myLora.initTTN("B827EBFFFE9D3258"); to line 66and get an error
TXfastViaTTN.ino: In function 'void setup();;TxfastViaTTN.ino:80class rn2483 has no member named 'initTTN'myLora.initTTN("B827ebfffe9d3258");
What do you think I have setup incorrectly.thank you Steven Kramer

gonzalocasas, 2016-07-14:
I am not sure which library you're using. If I google the function names, I assume it is `jpmeijers/ttnmapperarduino`, and just looking around quickly, I notice that the comment/documentation is outdated and that function indeed doesn't exists as such. I think it's been renamed, or the signature changed, but please check with the maintainer of that library. 

Alternatively, try the library from the TTN guys here: https://github.com/TheThingsNetwork/sdk/tree/master/devices/TheThingsUno