ttn-zh/ic880a-gateway #9: Important power

dicktonyboy, 2016-10-03:
What does this mean exactly?

Plug the iC880a (WARNING: first power plug to the wall socket, then to the gateway DC jack, and ONLY THEN USB to RPi!)

gonzalocasas, 2016-10-03:
This warning is copied from IMST's datasheet about the iC880A USB version:

[image]

Page 12 of this document: http://www.wireless-solutions.de/images/stories/downloads/Radio%20Modules/iC880A/iC880A_Datasheet_V0_17.pdf

dicktonyboy, 2016-10-07:
Thanks for this - I almost knew as I hit the button that it would be resolved.  Horribly ambiguous referring to a **now defunct version**. I see they have this on the current SPI version.  
[image]

http://www.wireless-solutions.de/images/stories/downloads/Radio%20Modules/iC880A/iC880A-SPI_QuickStartGuide.pdf

closed 2016-10-07 by dicktonyboy