ttn-zh/ic880a-gateway #10: downlink doesn't work.

byrann, 2016-12-06:
Hi,
I installed ic880a-gateway as shown and I am testing it using an arduino uno node with an SX1276.
Sending informations from my node to the TTN works fine but from ic880a-gateway to the connected device doesn't work, but i have to mention that in the "start.sh" code interface, the gateway seems to correctly send the message.
do I have to change my node ??
any advice could help!
thanks.

gonzalocasas, 2016-12-06:
Hi @byrann, I haven't heard so far that the downlink issues could be caused by the gateway installation. In general, this is an issue with the code. One that I've seen is that the LMiC library version was outdated, so after an update, the thing started to work. Could you please try [this very detailed walkthrough](http://www.tamberg.org/chopen/2016/LoRaWANIoTWorkshop.pdf) until the part of downlink and let me know? If you still have the problem, we'll need to analyze more in depth, but it's almost for sure a node-side issue.

Cheers

byrann, 2016-12-06:
thank you for replying @gonzalocasas.
I am going to follow that tuto and let you informed.

Cheers

byrann, 2016-12-07:
Hi, 
I followed the instructions in the tutorial and my problem is still there.
the "activation"  in node.js for my ABP app doesn't seem to work and about downlink, the device is still not recieving any message from the ic880a-gateway.

PS:I am using a dragino LoRa shield V1.1
Do you need more details or screenshots ?
@gonzalocasas can we be in touch through e-mail ? 
Cheers.

gonzalocasas, 2016-12-07:
Hi @byrann!

With 'activation' I guess you mean the activation event in a node.js MQTT client, right? In that case, it is normal that ABP doesn't trigger it, because there's no real activation going on, ABP is basically always activated.

The only additional thing regarding gateway config you might check is this page: http://staging.thethingsnetwork.org/gatewaystatus/ Make sure that your gateway EUI is listed there and both columns are marked as `active`.

But now that you mention the version of the dragino, I guess that is where the problem is. The first 3 iterations (1.0, 1.1 and 1.2) all had issues. Please try getting a 1.3 revision (or a different module altogether).

As for email, mine is githubuser at gmail, but I get notified of GH issues faster than email, and having the diagnosis in a public page might help others, so, I'd prefer keeping it here if possible.

byrann, 2016-12-07:
thank you very much @gonzalocasas.
Now all is clear for me :) .
I am going to get a1.3 revision of dragino lora shield for my tests; but speaking about modules supporting the TTN, is there any other sx1276 module supporting TTN but smaller than arduino shields??

PS: RN2483 are available only 01/03/2017.

Regards.

gonzalocasas, 2016-12-07:
There are many available actually, here are some options:
 - HopeRF RFM95 (directly, without shield): there's one option lab story on [how to build the cheapest node possible](https://www.thethingsnetwork.org/labs/story/build-the-cheapest-possible-node-yourself) with it. Or alternatively, get one of the many breakouts for it on tindie/oshpark/etc and avoid the wiring part.
  - RN2483: it is available on DigiKey on next-day delivery. Also mouser has it on stock. Here's a [very nice site to check availabilities/prices of chips](http://www.findchips.com/search/rn2483).
  - RN2483 on a breakout: this is especially good if you don't want to solder it, because is really trick to get right. Here's one of those [on Tindie by DrAzzy](https://www.tindie.com/products/DrAzzy/lorawan-rn2483rn2903-breakout-board-assembled/?pt=full_prod_search).
  - LoPy: if you're into Python, that's a good option. It had issues on release, but I think they are ironing them out. There's a [draft guide on TTN's website](https://www.thethingsnetwork.org/docs/draft/lopy/) already.
  - IMST also produced modules, they are pretty good, but you need to be comfortable with STM32 programming.
  - Laird RM1xx: also an interesting module, with BLE on it and a funky SmartBASIC interpreter. [Link](http://www.lairdtech.com/products/rm1xx-lora-modules).

Hope this helps!

closed 2016-12-25 by gonzalocasas