ttn-zh/ic880a-gateway #11: 915MHz alternative?

CEOGavin, 2016-12-14:
Hi Guys,

The IMST iC880a board only supports 868MHz. Is there any way to have a 915MHz implementation (legal requirements here in Australia)? 

Please help. Thanks

gonzalocasas, 2016-12-14:
Hi @CEOGavin! IMST is not manufacturing a 915Mhz version, but RisingHF is, and the good news is that this script is compatible with it. RisingHF has even a board to adapt to the RPi directly, the only difference in that case, is that reset pin needs to be changed, but other than that, it should just work.

gonzalocasas, 2016-12-14:
I'm gonna close this one, but feel free to contact me on gonzalocasas at gmail

closed 2016-12-14 by gonzalocasas

thayuriko, 2017-08-04:
Hi @gonzalocasas, I'm using a RisingHF board with this adaptor board, but didn't quite understand what you meant with "the only difference in that case, is that reset pin needs to be changed, but other than that, it should just work.". Where should I change the reset pin?

gonzalocasas, 2017-08-06:
Hi @thayuriko, if you use RisingHF's adapter board, then the pin that is used to reset the concentrator is not the same as the one used on the IMST adapter boards. So, you need to change one line of the `start.sh` script to the correct pin number. The line is this one: https://github.com/ttn-zh/ic880a-gateway/blob/spi/start.sh#L4

I hope this clarifies the situation!

cheers!

vaz83, 2018-12-07:
what's the pin number in this situation? i am facing the same problem