ttn-zh/ic880a-gateway #12: Support for The Things Network Gateway Connector

alviso, 2016-12-25:
Can you guys update the script to support the Gateway Connector according to this
https://www.thethingsnetwork.org/docs/gateways/registration.html
Thanks, Peter

gonzalocasas, 2016-12-25:
@alviso: yes, we will definitely do so as soon as possible. But it means updating the packet forwarder with a version that uses the new TTN connector, so, it's not a quickie.

alviso, 2016-12-25:
@gonzalocasas Thanks, looking forward to it!

jpmeijers, 2017-02-21:
A beta version of the new gateway connector for Linux is now available at:
https://github.com/kersing/packet_forwarder/tree/master/mp_pkt_fwd

It will be great if you could update you repository to support this. Because after that we can also solve this issue:
https://github.com/rayozzie/ttn-resin-gateway-rpi/issues/2

cimm, 2018-07-05:
The Things Network [packet forwarder](https://github.com/TheThingsNetwork/packet_forwarder/tree/master) seems to be placed on hold. What is the recommended package forwarder to use now, the Semtech UDP one until future notice?

gonzalocasas, 2018-07-05:
You have two options right now: either the Semtech UDP one, or the [mp packet forwarder](https://github.com/kersing/packet_forwarder/tree/master/mp_pkt_fwd). 

In the future, the gateway connector will be used by the gateway agent to interact with the Semtech UDP package forwarder (once v3 is released).