ttn-zh/ic880a-gateway #13: poly_pkt_fwd no such file or directory

fabio-C, 2017-01-23:
I followed the instructions and when I launch start.sh I get:
`
./start.sh: line 43: ./poly_pkt_fwd: No such file or directory`

gonzalocasas, 2017-01-23:
Hi @fabio-C, something must be wrong because at no point are you supposed to lauch `start.sh` manually.
Why did you end up running it manually? Is the systemd process not running?

fabio-C, 2017-01-24:
After launch "install.sh" what is the next step? Does the gateway software start automatically?

gonzalocasas, 2017-01-24:
Yes. Install creates a systemd service, you can query the status of the service using normal systemd commands, the service is called `ttn-gateway`. You can also use journalctl to check logs of that specific unit (or just grep `syslog`)

gonzalocasas, 2017-06-16:
Closing until we get more details from @fabio-C

closed 2017-06-16 by gonzalocasas

Hitheshkaranth, 2018-05-15:
@gonzalocasas 
Greeting I'm encoutering the same problem.

gonzalocasas, 2018-05-15:
@Hitheshkaranth as I said above, you're not supposed to run start manually, it runs as a service so triggering it by hand will not work as expected.

On the other hand, your log has nothing to do with the one in the original comment. What makes you think it's the same?

Hitheshkaranth, 2018-05-15:
Greetings,

Thanks for replying. I am stuck with the concentrator not starting and has
no idea how to start it. Can you guide me how to set this up?

gonzalocasas, 2018-05-15:
Check if you really have SPI enabled
Check if the power supply is enough (at least 2A!)
Check if there's no hardware damage
Then check the proper logs, don't try running the service by hand, use
syslog or journalctl to see the log output of the service.

Hitheshkaranth, 2018-05-16:
> issuecomment-389051607
> GYdsrNEyt9lRJWpO79fJd5Xks5tym5IgaJpZM4LrGo8

I apologize for late reply,

1.Yes,SPI is enabled on my board
2.Yes,the power source is ample enough
3.Yes but no sign of LoRa gate way turning on I viewed it on journalctl no
sign of services related to lora.

ERROR: [main] failed to start the concentrator

every time I try to run it this seems to be the error thrown up.

gonzalocasas, 2018-05-20:
> 3.Yes but no sign of LoRa gate way turning on I viewed it on journalctl no sign of services related to lora.

If there's really no sign of ttn-gateway service on `journalctl` then you have a misconfiguration. Please start on a clean image and re-install to be sure. 

> every time I try to run it this seems to be the error thrown up.
As I said, if you're running it manually, something is wrong in your installation, because this runs as a service.