ttn-zh/ic880a-gateway #14: Install service file to correct location

dbrgn, 2017-02-01:
Currently the service file is installed to `/lib/systemd/system/`, but this directory is not meant for user defined services.

Custom services should be installed to `/etc/systemd/system/` instead.

gonzalocasas, 2017-02-01:
@dbrgn: oh very good point; after MakeZurich, I will try to change this without breaking existing installations :)

dbrgn, 2017-02-01:
/etc/systemd/system overrides /lib/systemd/system :) Or you could remove the one in /lib if it exists before creating the one in /etc.

dbrgn, 2017-06-01:
Actually to be perfectly correct, the target directory should be the value returned by `pkg-config systemd --variable=systemdsystemunitdir`: https://www.freedesktop.org/software/systemd/man/systemd.html#Directories