ttn-zh/ic880a-gateway #22: time receiving

YousseMes, 2017-05-30:
Hi !
thank you for your Job !
I have just one probleme why I receive date every minute and not in really time 
thears some configuration for fix that 

[image]

gonzalocasas, 2017-05-31:
Hi!
Could you please add a bit of context. I don't even know exactly what that screenshot is.
But in any case, true real time is not something that works well with LoRaWAN, although I assume you don't mean that but more the time it takes for a message to reach your application. Bit without knowing more about your implementation is very hard to say.

And in any case, chances are the issues at hand are not related to the gateway at all, gateways are pretty simple forwarders.

gonzalocasas, 2017-06-16:
I'm closing this until we get more details from @YousseMes .

closed 2017-06-16 by gonzalocasas