ttn-zh/ic880a-gateway #23: ttn-gateway service entered failed state

thayuriko, 2017-08-04:
Hey guys,
After installing the install.sh and rebooting, I wanted to check if the service was running (since it was not being shown in the console) and here is what I got:

> rxhf@radek-gw2:~/ic880a-gateway $ sudo service ttn-gateway status
> ● ttn-gateway.service - The Things Network Gateway
>    Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled)
>    Active: activating (auto-restart) (Result: exit-code) since Fri 2017-08-04 17:33:02 UTC; 4s ago
>   Process: 9538 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
>  Main PID: 9538 (code=exited, status=1/FAILURE)
> 
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: Beacon is disabled
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: Monitor is disabled
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: Contact email configured to "thais.midorikawa@gmail.com"
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: Description configured to "Radek Gateway 2"
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: Successfully contacted server router.eu.thethings...work
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: INFO: [main] Starting the concentrator
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: ERROR: Version of calibration firmware not expected, ac...ed:2
> Aug 04 17:33:02 radek-gw2 ttn-gateway[9538]: ERROR: [main] failed to start the concentrator
> Aug 04 17:33:02 radek-gw2 systemd[1]: ttn-gateway.service: main process exited, code=exited, status=...LURE
> Aug 04 17:33:02 radek-gw2 systemd[1]: Unit ttn-gateway.service entered failed state.
> Hint: Some lines were ellipsized, use -l to show in full.

I get this "failed to start the concentrator" quite often in my other gateway, but after retrying a few times the gps_packet_forward it works. Should I just retry with this one as well? If so, how? By rebooting it and rechecking the status of the service over and over?

gonzalocasas, 2017-08-06:
Hi @thayuriko, I wouldn't use the gps_packet_forwarder, it's not really tested much. The poly_packet_forwarder on the other hand, is used by the whole community, and it's been working pretty well... About the error message, the issue with `failed to start the concentrator` means that the SPI connection between the RPi and the concentrator is not working correctly. This can be for a multitude of reasons:
 - If the wiring is wrong, so, you would get this consistently every single time.
 - If, however, you have the adapter board instead of jumper wires between boards, the reset pin could be wrong. This can cause the error to appear mostly every time.
 - If the power supply unit is not delivering enough (make sure it's 2A, and a good quality one), you might also get this very often. If you get it intermittently, this might be the first to examine.

Cheers
gonzalo

thayuriko, 2017-08-07:
Thank you, Gonzalo. I changed the Reset Pin as you recommended and now it works! :)
For those who are struggling with the same thing (and using a RHF0M301 + RPi3 + adaptor board), check this https://github.com/ttn-zh/ic880a-gateway/issues/11#issuecomment-320494038

closed 2017-08-07 by thayuriko

vaz83, 2018-12-07:
what's the correct pin number ?