ttn-zh/ic880a-gateway #25: problem with installation due to latest release of raspian

rmeldo, 2017-08-30:
Hi I found a problem when running install.sh

It couldn't find eth0 on a raspberry pi 3

I found the fix in this link:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191639

You may want to consider mentioning it in the installation instructions, as it took me about 4 hours to diagnose and find the fix.

Thanks
Riccardo

gonzalocasas, 2017-08-31:
We will fix the installer very soon, but please note that you are using Debian Stretch, and the instructions are for Debian Jessie. In Stretch, network interface names have changed format, which causes the script to fail.

rmeldo, 2017-08-31:
Thanks.
I couldn't find jessie and the link sent to the stretch page on the raspberrypi site so I used that.

By the way, could it also be the reason why after  install.sh ran without errors when I launch start.sh with the command sudo ./start.sh it gives the following error: "./start.sh: line 55: ./poly_pkt_fwd: No such file or directory"?

Riccardo

gonzalocasas, 2017-08-31:
The ` No such file or directory` error looks like something else.. did you make changes to the script for it to work on Stretch? did you select the local or the remote config mode? If you chose the remote and your config file was not yet merged, then this can happen.

rmeldo, 2017-08-31:
No, no change at all to the install.sh script. In fact the only change was the fix I mentioned above. 
I rerun  install.sh watching the screen for errors or warnings but nothing appeared.

One thing to mention is that, although I edited visudo to add the line ttn ALL=(ALL) NOPASSWD: ALL
I am still required to enter my password when I SSH into the rpi. Is it supposed to happen?
However I don't quite see how this could cause the error in start.sh

Riccardo

rmeldo, 2017-08-31:
By the way,
I selected the local mode, using default names for the gateway (ttn-gateway and ic880a)

rmeldo, 2017-08-31:
Hi,
I've made some progress but I still haven't managed to get it working.
I have installed Jessie lite instead of Stretch to avoid ant version problems.
I also found out that the  poly_pkt_fwd not found error was due to the fact that I was launching the start.sh shell from the directory ic880a.
when Iaunched from the directory /opt/ttn-gateway/bin it works but the script stops with an error:
"Version of calibration firmware not expected, actual:0 expected:2"

Any suggestions on what to do?

Thanks
Riccardo

gonzalocasas, 2017-08-31:
Hi @rmeldo, you don't need to launch the `poly_pkt_fwd` by hand, the installer creates a systemd service, and the thing is running in the background on start. Check the status of the service `systemctl status ttn-gateway` first

rmeldo, 2017-09-01:
thanks.

I used the command "systemctl status ttn-gateway "and I get the following:

● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled)
   Active: active (running) since Fri 2017-09-01 08:20:14 BST; 716ms ago
 Main PID: 6035 (start.sh)
   CGroup: /system.slice/ttn-gateway.service
           ├─6035 /bin/bash /opt/ttn-gateway/bin/start.sh
           └─6054 ./poly_pkt_fwd

if I rerun the command straight away I get the following:

● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2017-09-01 08:21:31 BST; 2s ago
  Process: 6289 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
 Main PID: 6289 (code=exited, status=1/FAILURE)

It looks like poly_pkt_fwd is causing the error or something is causing it to.
the service tries to restart every 5 seconds and the errors repeat.

I get the same result when 
sudo tail -f /var/log/syslog

Sep  1 10:35:08 ttn-gateway ttn-gateway[2447]: ERROR: Version of calibration firmware not expected, actual:0 expected:2

Any advice?

Thanks
Riccardo

gonzalocasas, 2017-09-01:
`Version of calibration firmware not expected` roughly means the SPI communication is not working well, it could be a lot of things, make sure you have a 2A power supply, if you already have one, try a different one; if you used jumper wires, double and triple check them, ideally use a backplane instead.

rmeldo, 2017-09-01:
Hi
I Watched the video from imst and for the "lite gateway" ( The one i bought). The setup includes editing start.sh in the /opt/ttn-gateway/bin directory to change the pin number from 25 to 5 and comment the line after the first sleep 0.1, which deals with pin numbering as well.
After a reboot then the concentrator starts and the gateway connects.

Previously I had been following the instructions from another link by imst. 
I lost some time but at least I deliver a bit more into the software

Thanks for the advice.
Riccardo

gonzalocasas, 2018-02-04:
Fixed

closed 2018-02-04 by gonzalocasas