ttn-zh/ic880a-gateway #26: Gateway

udaikumar59, 2017-10-27:
Hi,
we are setting up a lora gateway and we followed the same steps mentioned in https://github.com/ttn-zh/ic880a-gateway. but we are unable to find complete procedure to run the gateway connected to thethings network. we will be glad to help us in resolving the issue.
Thanks in Advance!
Regards

gonzalocasas, 2017-10-27:
Please describe the error otherwise it's hard to say what didn't work

udaikumar59, 2017-10-30:
hi

Thanks for your reply. We had tried with all the possibilities to start the concentrator but another important thing is that our gateway status is shown connected and packets are transmitted from the node also received at the ttn. We just viewed the data by integrating the cayenne LLP and registered the devices in  mydevices website.

Attached log file 

when we started the device again after 2 days, now the device is not connecting to ttn which means the ttn shows that gateway is not connected. attached a screenshot of this also with the log file of sudo tail -f /var/log/syslog and service ttn-gateway status

elektroniklab logo for Signature (2)

Thanks and regards ,

T .udhaya Kumar GM ( Tecchnical )

elektroniklab pvt ltd 

10 A , Masilamani street,  T.Nagar, Chennai - 600017

Ph:91 44 24336953 Fax :91 4424322819  mobile : 91 9840613199

Skype : udai.kumar1 - web : www.elektroniklab.com <http://www.elektroniklab.com/> 

 

elabindia@ttn-gateway:/opt/ttn-gateway/packet_forwarder/poly_pkt_fwd $ sudo ta -f /var/log/syslog
Oct 30 08:23:53 ttn-gateway rsyslogd-2007: action ‘action 17’ suspended, next retry is Mon Oct 30 08:25:23 2017 [try http://www.rsyslog.com/e/2007 ]
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.644351] usb 1-1.4: New USB device found, idVendor=046d, idProduct=c534
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.644365] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.644373] usb 1-1.4: Product: USB Receiver
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.644381] usb 1-1.4: Manufacturer: Logitech
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.653000] input: Logitech USB Receiver as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:046D:C534.0003/input/input2
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.721037] hid-generic 0003:046D:C534.0003: input,hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-3f980000.usb-1.4/input0
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.728772] input: Logitech USB Receiver as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:046D:C534.0004/input/input3
Oct 30 08:23:53 ttn-gateway kernel: [ 8112.790561] hid-generic 0003:046D:C534.0004: input,hiddev0,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-3f980000.usb-1.4/input1
Oct 30 08:23:53 ttn-gateway systemd-udevd[2985]: failed to execute ‘/lib/udev/mtp-probe’ ‘mtp-probe /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4 1 5’: No such file or directory
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PUSH_DATA acknowledged: 0.00%
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ### [DOWNSTREAM] ###
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PULL_DATA sent: 3 (100.00% acknowledged)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # RF packets sent to concentrator: 0 (0 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # TX errors: 0
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ### [GPS] ###
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # Invalid gps time reference (age: 1509348124 sec)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # Manual GPS coordinates: latitude 13.02370, longitude 80.14300, altitude -1 m
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ##### END #####
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 149 ms
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 148 ms
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 149 ms
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ##### 2017-10-30 07:22:34 GMT #####
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ### [UPSTREAM] ###
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # RF packets received by concentrator: 0
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # RF packets forwarded: 0 (0 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PUSH_DATA datagrams sent: 1 (217 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PUSH_DATA acknowledged: 0.00%
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ### [DOWNSTREAM] ###
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PULL_DATA sent: 3 (100.00% acknowledged)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # RF packets sent to concentrator: 0 (0 bytes)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # TX errors: 0
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ### [GPS] ###
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # Invalid gps time reference (age: 1509348154 sec)
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: # Manual GPS coordinates: latitude 13.02370, longitude 80.14300, altitude -1 m
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: ##### END #####
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 148 ms
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 149 ms
Oct 30 08:24:34 ttn-gateway ttn-gateway[2564]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 149 ms
[... cut ...]
INFO: Using fake GPS coordinates instead of real.
INFO: Upstream data is enabled
INFO: Downstream data is enabled
INFO: Ghoststream data is disabled
INFO: Radiostream data is enabled
INFO: Statusstream data is enabled
INFO: Beacon is disabled
INFO: Monitor is disabled
INFO: Contact email configured to "elab.ttn@gmail.com"
INFO: Description configured to "elabindia"
INFO: Successfully contacted server router.eu.thethings.network
INFO: [main] Starting the concentrator
ERROR: [main] failed to start the concentrator

gonzalocasas, 2017-10-30:
If it was working and it started failing with `INFO: [main] Starting the concentrator - ERROR: [main] failed to start the concentrator` then something like the power supply might be faulty. Try replacing your PSU, making sure it's delivering stable output.

udaikumar59, 2017-10-30:
Thanks and lot will get back after trying

udaikumar59, 2017-11-01:
Hi

 

Thanks for the help , we changed the PSU & still the issue “failed to start the concentrator “ persists in the system. However the ttn network shows the connected status.

 

These are the problems, we found out in the system,

 

1.On a fresh day start, the ttn network shows connected status but last connected shows the previously connected time interval. So on every fresh day start, we reset the concentrator and make it live

2. There is some issue in lgw_connect() & lgw_hal_success() and that’s why the problem still persists.

3. when we run the test_loragw_hal() function it results “Impossible to start the concentrator”

4. when we tested test_loragw_spi(), it shows data received=48.

5. The data sent through program is ELAB and those data is not been received at the ttn end as payload.

6.Some raw data is been transmitted to the end on previous connect to ttn. Please let us know how to decrypt the data.

7.how to make the node to transmit the data or how to make the concentrator to receive the data. 

 

I followed https://github.com/matthijskooijman/arduino-lmic this page to makeup the node and even it shows these status

 

Starting…

Packet queued

178975: EV_TX COMPLETE (Waiting for RX Window)

Packet queued

 

8. Do we need to setup the pin configuration as per SX1301 PIN Configuration  given in https://github.com/Lora-net/lora_gateway/wiki/SX1301-GPIO-output-configuration

9. Pin configuration of imst and rpi as we connected in our style

 

Imst       rpi

13                                                           Reset                                                                                                    11

14

SPI CLK

23

15

MISO

21

16

MOSI

19

[... 132 lines cut ...]
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # PULL_DATA sent: 3 (100.00% acknowledged)
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # RF packets sent to concentrator: 0 (0 bytes)
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # TX errors: 0
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: ### [GPS] ###
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # Invalid gps time reference (age: 1509512559 sec)
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # Manual GPS coordinates: latitude 13.02370, longitude 80.14300, altitude -1 m
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: ##### END #####
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 161 ms
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 162 ms
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 161 ms
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: ##### 2017-11-01 05:03:09 GMT #####
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: ### [UPSTREAM] ###
Nov  1 06:03:10 ttn-gateway ttn-gateway[3323]: # RF packets received by concentrator: 0
-------------------------------------------------------------------------------------
elabindia@ttn-gateway:/opt/ttn-gateway/lora_gateway/libloragw $ sudo ./test_loragw_spi
Beginning of test for loragw_spi.c
data received (simple read): 48
End of test for loragw_spi.c
-------------------------------------------------------------------------------------
elabindia@ttn-gateway:/opt/ttn-gateway/lora_gateway/libloragw $ sudo ./test_loragw_hal -a 868 -b 867.5 -t 868 -r 1257 -k 1
Beginning of test for loragw_hal.c
*** Library version information ***

*** Impossible to start concentrator ***

gonzalocasas, 2018-02-04:
This issue is very confusingly mixing gateway and node issues. I will close it because I've retested the gateway installation and it works fine, so, the issue is not related to this software repository (it could be hardware, or other environmental condition).

closed 2018-02-04 by gonzalocasas