ttn-zh/ic880a-gateway #27: Gateway not connected

themaawaa, 2017-11-10:
Hello, as of last month I successfully connected my gateway to the network.
Since then however, I have not been able to successfully recreate this connection even though nothing has changed in the configuration.

Illustration:
[image]

The current gateway I am using is a ic880-gateway using the TTN legacy packet forwarder protocol. My gateway is (succesfully) registered using the documentation in the README.md of this repository.

You can find my remote config [here](https://github.com/ttn-zh/gateway-remote-config/blob/master/B827EBFFFED1428E.json)

gonzalocasas, 2017-11-10:
Hi @themaawaa , the link to the image doesn't work. Could you please post it again? Thx

themaawaa, 2017-11-10:
@gonzalocasas Hey gonzalo, for some reason marking my gateway as indoor resulted in it using a gateway far away from me resulting in massive packet loss. Resetting the pi and marking it as outdoor has solved the issue.

closed 2017-11-11 by gonzalocasas