ttn-zh/ic880a-gateway #29: This setup is not working anymore?

anyn99, 2017-12-01:
Hi there!
Am I right that this repository is not up-to-date and working with its resources?

I found that install.sh is referring to a binary called poly_pkt_fwd, but it seems that the code
at https://github.com/TheThingsNetwork/packet_forwarder doesn't compile to a binary callled poly_pkt_fwd (anymore?).

What would be the current approach if i want to build a gateway with RPi and ic880a?
I guess the stuff at https://github.com/TheThingsNetwork/lora_gateway is still usable?
What packet forwarder is the best choice atm?

Thanks in advance for any hints!

abouillot, 2017-12-01:
The repo is complete. I've set up a GW last week with it. follow the wiki instructions. The binaries are compiled during the process
https://github.com/ttn-zh/ic880a-gateway/wiki

gonzalocasas, 2017-12-01:
@anyn99: Yes, @abouillot is correct. This repository points to a specific branch of the packet_forwarder repo that does have the binary required.

closed 2017-12-01 by gonzalocasas

anyn99, 2017-12-01:
Oh, i overlooked that the "legacy" branch is used.
Sorry for the hassle :-)