ttn-zh/ic880a-gateway #33: ERROR: [main] failed to start the concentrator

safurniss, 2018-02-14:
I've just received my IC880a gateway module and built it following the guide at https://github.com/ttn-zh/ic880a-gateway/wiki.

I have tripple check all of the jumper connections and they match the pin-out from the article.

However I don't see my gateway as connected in the TTN dashboard.

I've run the following commands to try and troubleshoot the issue:-

ttn@ttn-gateway:~ $ service ttn-gateway status
● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-02-14 22:10:27 GMT; 343ms ago
 Main PID: 4665 (start.sh)
   CGroup: /system.slice/ttn-gateway.service
           ├─4665 /bin/bash /opt/ttn-gateway/bin/start.sh
           └─4676 sleep 0.1

ttn@ttn-gateway:~ $ sudo tail -f /var/log/syslog
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Statusstream data is enabled
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Beacon is disabled
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Monitor is disabled
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Contact email configured to "MY_EMAIL_ADDRESS"
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Description configured to "TTN"
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: Successfully contacted server router.eu.thethings.network
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: INFO: [main] Starting the concentrator
Feb 14 22:12:21 ttn-gateway ttn-gateway[5287]: ERROR: [main] failed to start the concentrator
Feb 14 22:12:21 ttn-gateway systemd[1]: ttn-gateway.service: Unit entered failed state.
Feb 14 22:12:21 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
Feb 14 22:12:26 ttn-gateway systemd[1]: ttn-gateway.service: Service hold-off time over, scheduling restart.
Feb 14 22:12:26 ttn-gateway systemd[1]: Stopped The Things Network Gateway.
Feb 14 22:12:26 ttn-gateway systemd[1]: Started The Things Network Gateway.
Feb 14 22:12:27 ttn-gateway ttn-gateway[5333]: /opt/ttn-gateway/gateway-remote-config /opt/ttn-gateway/bin
Feb 14 22:12:28 ttn-gateway ttn-gateway[5333]: Already up-to-date.
Feb 14 22:12:28 ttn-gateway ttn-gateway[5333]: HEAD is now at 8fd2594 Merge pull request #568 from zara-lab/master
Feb 14 22:12:28 ttn-gateway ttn-gateway[5333]: /opt/ttn-gateway/bin
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: [TTN Gateway]: Use Gateway EUI B827EBFFFE7CB189 based on eth0
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: *** Poly Packet Forwarder for Lora Gateway ***
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: Version: 2.1.0
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: *** Lora concentrator HAL library version info ***
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: Version: 3.1.0; Options: native;
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: ***
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Little endian host
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: found global configuration file global_conf.json, parsing it
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: lorawan_public 1, clksrc 1
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Configuring TX LUT with 16 indexes
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: gateway MAC address is configured to AA555A0000000000
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Found 2 servers in array.
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Server 0 configured to "127.0.0.1", with port up "1680" and port down "1681"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Server 1 configured to "iot.semtech.com", with port up "1680" and port down "1680"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Found 5 system calls in array.
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: System command 0: "df -m"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: System command 1: "free -h"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: System command 2: "uptime"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: System command 3: "who -a"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: System command 4: "uname -a"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: monitor hostname or IP address is configured to "127.0.0.1"
[... 43 lines cut ...]
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Reference altitude is configured to 115 meters
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: GPS is enabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: antenna_gain 0 dBi
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Upstream data is enabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Downstream data is enabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Ghoststream data is disabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Radiostream data is enabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Statusstream data is enabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Beacon is disabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Monitor is disabled
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Contact email configured to "safurniss@blueyonder.co.uk"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Description configured to "TTN"
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: Successfully contacted server router.eu.thethings.network
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: [main] Starting the concentrator
Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: ERROR: [main] failed to start the concentrator
Feb 14 22:12:29 ttn-gateway systemd[1]: ttn-gateway.service: Unit entered failed state.
Feb 14 22:12:29 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.

I'm running the latest Raspbian Stretch Lite build and the Pi connects by LAN cable.

Do you have any suggestions on what I should check?

Regards

Steve

gonzalocasas, 2018-02-15:
mmmm... hard to tell, maybe a silly question but, did you check if SPI is enabled?

safurniss, 2018-02-16:
Hi, yes SPI is enabled....... I have just swapped the Pi2 for a different Pi2 and did a fresh install using the local file not the remote file that was used in the previous install.... all is now working correctly.....

I've submitted a new remote .json file for the new Pi2 and will look at restarting once it has been committed to the repository to see if it was anything to do with that.... also will look at testing the other Pi2 that I couldn't get it working on to see if that was the problem.

gonzalocasas, 2018-02-16:
Ah, yes, I think if you run the installer and it pulls the remote repo
before the pull request has been merged, it will create a symlink for a
file that doesn't yet exist and then the next pull will fail.

MartinLavicka, 2018-05-24:
I connected reset (pin13) to pin11 on Raspberry pi 3 (GPIO0).

gonzalocasas, 2018-05-24:
@MartinLavicka why? Unless you want to manually change the installer, the instructions indicate using pin 22 for reset (BCM 25): https://github.com/ttn-zh/ic880a-gateway/wiki

Outfardine, 2018-07-06:
hi @safurniss, please i'm dealing with the same problem. can you tell me what should i do ?!

gonzalocasas, 2018-07-06:
This error message only says that the Raspberry Pi cannot connect to the concentrator board, you have to provide additional details to understand what is your concrete problem. Maybe SPI is disabled, maybe your power supply is too weak for the boards, maybe you selected remote config but didn't have the remote config file pushed to the repository....

Taher1322, 2019-01-08:
> This error message only says that the Raspberry Pi cannot connect to the concentrator board, you have to provide additional details to understand what is your concrete problem. Maybe SPI is [...]

I am using 5V,1.6A adapter, my remote config file is pushed to the repository, I have edited the reset code into my start.sh file, my SPI is enabled as well.

#! /bin/bash

# Reset iC880a PIN
SX1301_RESET_BOARD_PIN=22
echo "$SX1301_RESET_BOARD_PIN_22" > /sys/class/gpio/export 
echo "out" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/direction 
echo "0" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value 
sleep 0.1
echo "1" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value 
sleep 0.1
echo "0" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value
sleep 0.1
echo "$SX1301_RESET_BOARD_PIN_22" > /sys/class/gpio/unexport

# Test the connection, wait if needed.

After restating it, and checking the gateway status by running 
sudo service ttn-gateway status
 
I get 
● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor pres
   Active: active (running) since Tue 2019-01-08 09:40:48 GMT; 1s ago
 Main PID: 1610 (start.sh)
   CGroup: /system.slice/ttn-gateway.service
           ├─1610 /bin/bash /opt/ttn-gateway/bin/start.sh
           ├─1614 git pull
           ├─1615 git fetch --update-head-ok
           └─1616 git-remote-https origin https://github.com/ttn-zh/gateway-remo

Jan 08 09:40:48 ttn-gateway systemd[1]: Started The Things Network Gateway.
Jan 08 09:40:48 ttn-gateway ttn-gateway[1610]: /opt/ttn-gateway/gateway-remote-c
lines 1-12/12 (END)

When i run sudo tail -f /var/log/syslog 

I get failed to connect to concentrator. 

the status on ttn i still non connected.

Please help me with this issue..

waynemcl, 2019-04-07:
Is the line "Feb 14 22:12:29 ttn-gateway ttn-gateway[5333]: INFO: local_conf.json does not contain a JSON object named SX1301_conf" an issue for you?

alexxtasi, 2019-05-23:
Hi all
same problem with my gateway...
- having followed the instructions
- having SPI enabled
- running ```sudo ./install.sh spi```
- I don't use a remote setting file so I provide hostname etc, answering script's questions
the script perform RPi restart and after that, the ```ttn-gateway.service```  results in : 
```
● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-05-23 23:26:05 EEST; 3s ago
  Process: 5393 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
 Main PID: 5393 (code=exited, status=1/FAILURE)

May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: Beacon is disabled
May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: Monitor is disabled
May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: Contact email configured to "my@email"
May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: Description configured to "customHostname"
May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: Successfully contacted server router.eu.thethings.network
May 23 23:26:05 customHostname ttn-gateway[5393]: INFO: [main] Starting the concentrator
May 23 23:26:05 customHostname ttn-gateway[5393]: ERROR: [main] failed to start the concentrator
May 23 23:26:05 customHostname systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
May 23 23:26:05 customHostname systemd[1]: ttn-gateway.service: Unit entered failed state.
May 23 23:26:05 customHostname systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
```

this comes again and again after restarting the service or even RPi !!!

```/var/log/syslog``` shows : 
```
May 23 23:18:01 customHostname systemd[1]: ttn-gateway.service: Service hold-off time over, scheduling restart.
May 23 23:18:01 customHostname systemd[1]: Stopped The Things Network Gateway.
May 23 23:18:01 customHostname systemd[1]: Started The Things Network Gateway.
May 23 23:18:01 customHostname ttn-gateway[3505]: *** Poly Packet Forwarder for Lora Gateway ***
May 23 23:18:01 customHostname ttn-gateway[3505]: Version: 2.1.0
May 23 23:18:01 customHostname ttn-gateway[3505]: *** Lora concentrator HAL library version info ***
May 23 23:18:01 customHostname ttn-gateway[3505]: Version: 3.1.0; Options: native;
May 23 23:18:01 customHostname ttn-gateway[3505]: ***
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Little endian host
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: found global configuration file global_conf.json, parsing it
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: lorawan_public 1, clksrc 1
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Configuring TX LUT with 16 indexes
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: gateway MAC address is configured to AA555A0000000000
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Found 2 servers in array.
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Server 0 configured to "127.0.0.1", with port up "1680" and port down "1681"
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Server 1 configured to "iot.semtech.com", with port up "1680" and port down "1680"
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: Found 5 system calls in array.
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: System command 0: "df -m"
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: System command 1: "free -h"
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: System command 2: "uptime"
[... cut ...]
May 23 23:18:01 customHostname ttn-gateway[3505]: INFO: [main] Starting the concentrator
May 23 23:18:01 customHostname ttn-gateway[3505]: ERROR: [main] failed to start the concentrator
May 23 23:18:01 customHostname systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
May 23 23:18:01 customHostname systemd[1]: ttn-gateway.service: Unit entered failed state.
May 23 23:18:01 customHostname systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
```
any suggestions on what the problem is ??

regards

gonzalocasas, 2019-05-23:
The message means the RPi cannot communicate with the iC880a. The root cause can be a lot of things: 1) the wiring might be wrong (if you are using jumper wires, this is very often the case, even if you double checked), 2) the power supply is not sufficient (make sure you have a 2A unit at least, and of a good manufacturer), 3) the hardware might have a defect...

mojtaba-esk, 2020-08-26:
I had the same issue, I could resolve it by resetting the CE0 and RESET pin before running the forwarder. Here is how i did it:

```#!/bin/bash

echo "Restarting the CE0 pin..."
CSPIN=8
echo "$CSPIN" > /sys/class/gpio/export
sleep 1
echo "out" > /sys/class/gpio/gpio$CSPIN/direction
echo "0" > /sys/class/gpio/gpio$CSPIN/value
sleep 1
echo "1" > /sys/class/gpio/gpio$CSPIN/value
echo "Done\n"
sleep 1

echo "Restarting the Concentrator..."
RSTPIN=17
echo "$RSTPIN" > /sys/class/gpio/export
sleep 1
echo "out" > /sys/class/gpio/gpio$RSTPIN/direction
echo "1" > /sys/class/gpio/gpio$RSTPIN/value
sleep 1
echo "0" > /sys/class/gpio/gpio$RSTPIN/value
echo "Done\nLaunching the forwarder..."
sleep 2

cd /PATH_TO_THE_GATEWAY_CODES/packet_forwarder/ && ./lora_pkt_fwd```

Save this code in a shell file, change the path according to your own settings and make sure to run it with `sudo`
The full article can be found [here](https://www.mojiot.com/blog/resolved-error-main-failed-to-start-the-concentrator/)

wong-hao, 2020-10-04:
The code I used can be found at https://gitee.com/rejeee/gw1302s, plz.