ttn-zh/ic880a-gateway #34: GPS working but not syncing?

duineuk, 2018-03-10:
Morning all,

I've spent some time trying to resolve this, and discussed on the TTN forums. I'm beginning to suspect there may be a code issue.

I did a clean install of Stretch and got the GPS module set up and working first. It's a Raspi3 so I reconfigured the serial ports to feed the NMEA to the hardware UART. That gets me reliable position information on ttyAMA0:

...
$GPRMC,225036.000,A,51xx.xxxx,N,000xx.xx,W,0.00,200.38,090318,A*71
...

I then installed and configured ttn-zh per the wiki. Everything is up and running and the concentrator appears to be seeing the GPS:

...
Mar 9 22:47:09 ttn-dev-gw ttn-gateway[538]: INFO: Reference longitude is configured to 0.000000 deg
Mar 9 22:47:09 ttn-dev-gw ttn-gateway[538]: INFO: Reference altitude is configured to 0 meters
Mar 9 22:47:09 ttn-dev-gw ttn-gateway[538]: INFO: GPS is enabled
...

(I say ‘appears’… it isn’t throwing any errors about opening the serial port. I’m slightly suspicious that I can still cat /ttyAMA0 in another shell.) 

However, the gps time reference is defaulting to unix epoch. It looks like perhaps an initial sync was missing:
...
Mar 9 22:49:32 ttn-dev-gw ttn-gateway[538]: ### [GPS] ###
Mar 9 22:49:32 ttn-dev-gw ttn-gateway[538]: # Invalid gps time reference (age: 1520635659 sec)
Mar 9 22:49:32 ttn-dev-gw ttn-gateway[538]: # Manual GPS coordinates: latitude 0.00000, longitude 0.00000, altitude 0 m
Mar 9 22:49:32 ttn-dev-gw ttn-gateway[538]: ##### END #####
...

Per the example above, the timestamp on each sentence format is correct. I’ve also patched the PPS output from the GPS module to pin 19 on the iC880a concentrator, although I don't think this matters.

I've had a look at poly_pkt_fwd.c and there's nothing obvious I can find, although this suggests there may be some work remaining:

                printf("### [GPS] ###\n");
                //TODO: this is not symmetrical. time can also be derived from other sources, fix
                if (gps_enabled == true) {

and:

                //TODO: is this okay, can time be recruited from the local system if gps is not working?
                /* get a copy of GPS time reference (avoid 1 mutex per packet) */
                if ((nb_pkt > 0) && (gps_active == true)) {
                        pthread_mutex_lock(&mx_timeref);
                        ref_ok = gps_ref_valid;
                        local_ref = time_reference_gps;
                        pthread_mutex_unlock(&mx_timeref);
                } else {
                        ref_ok = false;

...although by my reading that suggests alternate sources should only be used if the gps isn't working.

Thoughts?

gonzalocasas, 2018-03-10:
Hi @duineuk, I haven't tried this, but if i recall correctly, there're others that did. In any case, if there's a code issue, it is most likely on the poly packet forwarder, not on this installer.

duineuk, 2018-03-10:
Ah - apologies if this is the wrong place to post!

gonzalocasas, 2018-03-17:
Hi @duineuk, have you tried the instructions at the bottom of this page? https://sandboxelectronics.com/?p=2696
They highlight an issue with serial port on RPi that might be the same one you're having...?

duineuk, 2018-03-17:
Hi there. I do know about that funny, and manually set the unit up to disable BT and run the GPS into then hardware UART. 

I'm now running a container under resin.io which works fine - never did get this resolved, but can probably close it.

closed 2018-03-17 by duineuk

OhHeyAlan, 2018-04-26:
@duineuk Try setting `GW_FAKE_GPS` to `false` and `GW_GPS` to `true` under "Device Environment Variables".