ttn-zh/ic880a-gateway #36: Can I rename the gateway name

pierrot10, 2018-05-19:
Hello,
Thank for sending the black pane. I received it satureday.
I just configured my second gateway. For now it will remind at home close to my first gateway.

During the installation, I answered yes for the remote configuration and at the end the gateway was renamed to ttn-gateway. However, now I have two gateway with the same name, name I switched of the first.

I red your doc and you wrote

> By default, the installer changes the hostname of your Raspeberry Pi to ttn-gateway (to prevent collisions with other Raspberry Pis in your network). You can override this in non-remote configuration mode.

As I chosed remote configuration, I should not rename it? What happen if I do it, because I can not have two devise on the same network. Why should not it be renamed for remote configuration?

Thank for your reply.
Pierre

gonzalocasas, 2018-05-19:
You're free to rename it manually after the installation, that's no problem at all.

gonzalocasas, 2018-05-19:
It's not that it must not be renamed for remote, it's only that the installer doesn't give you the option to choose a name during remote install mode; but no technical reason for that, just not implemented.

So, is it ok if I close this one?

pierrot10, 2018-05-20:
Hello, Great thank!! 
Yesterday, I created a new gateway-remote-config file wit the uei B827EBFFFE46816E
Have a new week-end

closed 2018-05-20 by pierrot10