ttn-zh/ic880a-gateway #40: can't connect without any failure

elBeato, 2018-08-13:
Hi there, 
I've tried every thing what i found on google...but I'can't connect to the ttn network. this is my status and syslog: 
ttn@ttn-gateway:~ $ service ttn-gateway status
* ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-08-13 19:56:52 CEST; 58min ago
 Main PID: 211 (start.sh)
   CGroup: /system.slice/ttn-gateway.service
           |-211 /bin/bash /opt/ttn-gateway/bin/start.sh
           `-543 ./poly_pkt_fwd
ttn@ttn-gateway:~ $ sudo tail -f /var/log/syslog
[sudo] Passwort f�r ttn:
Aug 13 20:55:20 ttn-gateway systemd[911]: Listening on GnuPG cryptographic agent and passphrase cache.
Aug 13 20:55:20 ttn-gateway systemd[911]: Reached target Paths.
Aug 13 20:55:20 ttn-gateway systemd[911]: Listening on GnuPG cryptographic agent (access for web browsers).
Aug 13 20:55:20 ttn-gateway systemd[911]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Aug 13 20:55:20 ttn-gateway systemd[911]: Reached target Sockets.
Aug 13 20:55:20 ttn-gateway systemd[911]: Reached target Timers.
Aug 13 20:55:20 ttn-gateway systemd[911]: Reached target Basic System.
Aug 13 20:55:20 ttn-gateway systemd[911]: Reached target Default.
Aug 13 20:55:20 ttn-gateway systemd[911]: Startup finished in 180ms.
Aug 13 20:55:20 ttn-gateway systemd[1]: Started User Manager for UID 1001.
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA datagrams sent: 2 (444 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA acknowledged: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [DOWNSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_DATA sent: 4 (0.00% acknowledged)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # TX errors: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [GPS] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Invalid gps time reference (age: 1534186326 sec)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### END #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### 2018-08-13 18:52:36 GMT #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [UPSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets received by concentrator: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets forwarded: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA datagrams sent: 2 (444 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA acknowledged: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [DOWNSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_DATA sent: 6 (0.00% acknowledged)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # TX errors: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [GPS] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Invalid gps time reference (age: 1534186356 sec)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### END #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### 2018-08-13 18:53:06 GMT #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [UPSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets received by concentrator: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets forwarded: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA datagrams sent: 2 (444 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA acknowledged: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [DOWNSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_DATA sent: 6 (0.00% acknowledged)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # TX errors: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [GPS] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Invalid gps time reference (age: 1534186386 sec)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### END #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### 2018-08-13 18:53:36 GMT #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [UPSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets received by concentrator: 1
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets forwarded: 0 (0 bytes)
[... 41 lines cut ...]
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### END #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ##### 2018-08-13 18:55:06 GMT #####
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [UPSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets received by concentrator: 1
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets forwarded: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA datagrams sent: 2 (444 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PUSH_DATA acknowledged: 0.00%
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [DOWNSTREAM] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_DATA sent: 6 (0.00% acknowledged)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # TX errors: 0
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: ### [GPS] ###
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Invalid gps time reference (age: 1534186506 sec)
Aug 13 20:55:36 ttn-gateway ttn-gateway[211]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m

SPI is enable.
Power supply is original from RPI (V2.0)
I connect the IC880a with the backplane and an raspberry pi 2.0 Version B

Could anybody help me? 
kindly regards
Elbeato

elBeato, 2018-08-13:
oh...I've already found out, that I use a RPI 1 Model B+ V1.2, I think, this could be the problem??

elBeato, 2018-08-20:
Thank you for checking the log file!
ok, you can close this, but the Gateway doesn't show as connected in the console. I've done a correctly registration on the TTN console. I will try it today with the RPI 3...

gonzalocasas, 2018-08-20:
@elBeato ok, the first thing to check after installation would be that messages from a node are actually being forwarded. If that works, then the problem is definitely only a registration issue, in which case I would suggest to post it on the forum or on TTN's Slack. If messages are not forwarded, then maybe re-open it, adding the new, full log and any additional info that could help diagnose it.

closed 2018-08-20 by gonzalocasas