ttn-zh/ic880a-gateway #42: No downlink since update

rob-vandam, 2018-11-13:
I had issues with the gateway, so I did an update. The issue was that the gateway stopped forwarding packets after 4 days, and showed offline in the TTN backoffice. 

After the update I do not receive any downlink messages anymore. In de TTN backoffice I can see in the traffic overview of the app that the downlink message is received by TTN, but no data on both of my nodes. These nodes worked fine before the update.

Uplink data seems to work fine.

Os = 9 (stretch), raspbian

gonzalocasas, 2018-11-13:
These installer hasn't had changes in a long while, i guess the issue is more on the board's side, maybe a defective power supply unit or similar. I'll close the issue for now, but feel free to comment on it again.

closed 2018-11-13 by gonzalocasas

aparcar, 2019-02-15:
I'm having the same issue, trying OTAA but the nodes never receive the downlink messages. Did you solve the problem @rob-vandam ?

rob-vandam, 2019-02-15:
@aparcar the gateway works fine now. I did two things: at first I did an update (even if there was none available), and second I did open a few ports on the router: UDP 1700, TCP 1900 and1901, TCP 1883, TCP 8883

aparcar, 2019-02-16:
Works, thanks. It was a node related problem.