ttn-zh/ic880a-gateway #43: Select LoRaWAN network type upon installation

omogenot, 2018-12-12:
Hi,

First of all, thanks for this nice work.

I have though a request to extend the installation script to ask whether the gateway will be a LoRaWAN public or private network. One more question like "Do you use LoRaWAN private network? [y/N]" in the script...
This information is in the global_config.json file in the "SX1301_conf" section as :
"lorawan_public": true,

I don't know if this information can be replaced in the local_config.json.
It took me the whole day to figure out that my system was not working reliably because my endpoints are for private network and the default values this parameter is true ! If adding this question can save other people hours of research....

I thank you in advance,
Best regards,
OM.

gonzalocasas, 2018-12-18:
@omogenot it's an interesting idea. I haven't used private networks, since we want to promote the public TTN one, but I agree with you that having the option in the installer would probably be very helpful. Could you please send a pull request implementing the change? Thanks!

omogenot, 2018-12-18:
Hi,
Having never used the pull request of GitHub, I think it would take me hours to go through and be using a sledgehammer to crack a nut, whereas you are probably an expert in this domain, and "just" adding the question and the corresponding sed command in the install script would "just" take 5 minutes for you, isn't it?  ;-)

gonzalocasas, 2018-12-18:
@omogenot well, no, not really 😃

On one hand, using pull requests is something that is worth learning anyway, because you can contribute back to the open source community (either this project or any other).

On the other hand, if you have a feature request and have the knowledge to implement it (which you definitely have, since you basically spelled out the implementation right there in your comment), then it would be really nice of you if you would set aside a bit of your time and contribute back to the project.

This is all volunteering work, I don't get pay to do any of this, so, if we all put in a bit of our time, the project moves forward much faster and better, if not, then basically it moves forward only whenever I find the time to do things, and since this feature is really one that does not foster the public community network -quite the contrary-, what you're asking me is effectively if I can work for you for free, which is a tad unfair if you think about it.

omogenot, 2018-12-18:
Too philosophical for me :-D
I'll see as well when and if I have some free time...

closed 2018-12-18 by omogenot