ttn-zh/ic880a-gateway #45: Error: Failed to start the concentrator...

Taher1322, 2019-01-08:
I am using 5V,1.6A adapter, my remote config file is pushed to the repository, I have edited the reset code into my start.sh file, my SPI is enabled as well.

#! /bin/bash

Reset iC880a PIN
SX1301_RESET_BOARD_PIN=22
echo "$SX1301_RESET_BOARD_PIN_22" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/direction
echo "0" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value
sleep 0.1
echo "1" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value
sleep 0.1
echo "0" > /sys/class/gpio/gpio$SX1301_RESET_BOARD_PIN_22/value
sleep 0.1
echo "$SX1301_RESET_BOARD_PIN_22" > /sys/class/gpio/unexport

Test the connection, wait if needed.
After restating it, and checking the gateway status by running
sudo service ttn-gateway status

I get
● ttn-gateway.service - The Things Network Gateway
Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor pres
Active: active (running) since Tue 2019-01-08 09:40:48 GMT; 1s ago
Main PID: 1610 (start.sh)
CGroup: /system.slice/ttn-gateway.service
├─1610 /bin/bash /opt/ttn-gateway/bin/start.sh
├─1614 git pull
├─1615 git fetch --update-head-ok
└─1616 git-remote-https origin https://github.com/ttn-zh/gateway-remo

Jan 08 09:40:48 ttn-gateway systemd[1]: Started The Things Network Gateway.
Jan 08 09:40:48 ttn-gateway ttn-gateway[1610]: /opt/ttn-gateway/gateway-remote-c
lines 1-12/12 (END)

When i run sudo tail -f /var/log/syslog

Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: Beacon is disabled
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: Monitor is disabled
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: Contact email configured to "ujjainwala.taher1322@gmail.com"
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: Description configured to "Faclon_TTN_Gateway"
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: Successfully contacted server router.eu.thethings.network
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: INFO: [main] Starting the concentrator
Jan  8 10:09:55 ttn-gateway ttn-gateway[4693]: ERROR: [main] failed to start the concentrator
Jan  8 10:09:55 ttn-gateway systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 10:09:55 ttn-gateway systemd[1]: ttn-gateway.service: Unit entered failed state.
Jan  8 10:09:55 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
Jan  8 10:10:00 ttn-gateway systemd[1]: ttn-gateway.service: Service hold-off time over, scheduling restart.
Jan  8 10:10:00 ttn-gateway systemd[1]: Stopped The Things Network Gateway.
Jan  8 10:10:00 ttn-gateway systemd[1]: Started The Things Network Gateway.
Jan  8 10:10:00 ttn-gateway ttn-gateway[4722]: /opt/ttn-gateway/gateway-remote-config /opt/ttn-gateway/bin
Jan  8 10:10:05 ttn-gateway ttn-gateway[4722]: Already up-to-date.
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: HEAD is now at f031abf Merge pull request #1167 from Taher1322/master
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: /opt/ttn-gateway/bin
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: [TTN Gateway]: Use Gateway EUI B827EBFFFE0B103C based on wlan0
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: *** Poly Packet Forwarder for Lora Gateway ***
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: Version: 2.1.0
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: *** Lora concentrator HAL library version info ***
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: Version: 3.1.0; Options: ftdi;
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: ***
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: Little endian host
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: found global configuration file global_conf.json, parsing it
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: lorawan_public 1, clksrc 1
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: Configuring TX LUT with 16 indexes
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Jan  8 10:10:06 ttn-gateway ttn-gateway[4722]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
[... 58 lines cut ...]
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: packets received with no CRC will NOT be forwarded
Jan  8 10:10:07 ttn-gateway systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Reference latitude is configured to 19.117190 deg
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Reference longitude is configured to 72.887140 deg
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Reference altitude is configured to 16 meters
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: GPS is enabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: antenna_gain 0 dBi
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Upstream data is enabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Downstream data is enabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Ghoststream data is disabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Radiostream data is enabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Statusstream data is enabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Beacon is disabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Monitor is disabled
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Contact email configured to "ujjainwala.taher1322@gmail.com"
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Description configured to "Faclon_TTN_Gateway"
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: Successfully contacted server router.eu.thethings.network
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: INFO: [main] Starting the concentrator
Jan  8 10:10:07 ttn-gateway ttn-gateway[4722]: ERROR: [main] failed to start the concentrator
Jan  8 10:10:07 ttn-gateway systemd[1]: ttn-gateway.service: Unit entered failed state.
Jan  8 10:10:07 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.

The status on TTN console is still non connected.

Please help me with this issue..

Taher1322, 2019-01-12:
I used a 4A adapter and still i get the same issue, failed to start the concentrator...
Please help me regarding this issue

gonzalocasas, 2019-01-12:
1. Why did you edit the `start.sh` file?
2. What's your setup? (i.e. IMST board? RAK board?)
3. The edited file has several mistakes, chief among them is the fact that you declare the variable as `SX1301_RESET_BOARD_PIN` but then you use it as `SX1301_RESET_BOARD_PIN_22`. There's also a missing comment char at start of line 3. 

Please check all that before. The setup has been working without issues for several years, and unless something has changed recently, this is a mistake in the configuration, not a bug in the installer.

gonzalocasas, 2019-01-13:
If you use the iC880a concentrator board, then why did you change the `start.sh`? Assuming you followed the connection instructions, the script should work without changes.