ttn-zh/ic880a-gateway #51: My gateway stop transmiting

pierrot10, 2019-04-26:
Hello
I installed and configured two gateway following that procedure. Wedenesady, I stopped my gateway and when later I restated it, no way to be seen by TTN.
I have a second and same gateway and this one can be seen by TTN but now message are transmitted to TTN throught the second gateway.

Regarding the first gateway, which stop working, I moved iit to my place and I switch it on. Once again, from my flat, the gateway can not be seen by TTN.

I am planning to reinstalled it, but first there is a way (a command from Raspberry) to check the gateway (concentrator, antenna, application, etc...)?

Note that my Raspberry connected to the concentrator, starts well. I can be remotely connected to it, and I have an internet access. In spite of this, the gate way is not seen by TTN. Can it be possible that [Gateway remote configuration file] (https://github.com/ttn-zh/gateway-remote-config) is corrupt?

Thank
Cheers

gonzalocasas, 2019-04-26:
Please check the output of the logfiles (you can query it in different ways, one is `sudo journalctl --unit=ttn-gateway`). The most likely issue is that the concentrator board is somehow no longer able to start. It could be a hardware failure or a electric connectivity issue on the interface between the two boards (RPi and iC880a).

pierrot10, 2019-04-26:
Hello, thank for your reply. I planned to debug that issue, this evening. I will try and update the status

pierrot10, 2019-04-26:
Hello, I have some error. But first, there is way to no the EUI key. I have two gateway. If I look in /otp/ttn-gateway/gateway-remote-config/ there is so many different file between mine and I stupidely not recorded it

I just switch on my gateway, and I have soem error as the following

> error: object file .git/objects/62/002a0d748868181c53398b18c9af33c6dac128 is empty
Apr 26 18:28:17 smartsensors ttn-gateway[328]: error: object file .git/objects/62/002a0d748868181c53398b18c9af33c6dac128 is empty
Apr 26 18:28:17 smartsensors ttn-gateway[328]: fatal: loose object 62002a0d748868181c53398b18c9af33c6dac128 (stored in .git/objects/62/002a0d748868181c533
Apr 26 18:28:17 smartsensors ttn-gateway[328]: fatal: The remote end hung up unexpectedly
Apr 26 18:28:17 smartsensors ttn-gateway[328]: error: object file .git/objects/4a/d054cb435c7091efed8e5d8ba20b3e3ab5231d is empty
Apr 26 18:28:17 smartsensors ttn-gateway[328]: error: object file .git/objects/4a/d054cb435c7091efed8e5d8ba20b3e3ab5231d is empty
Apr 26 18:28:17 smartsensors ttn-gateway[328]: fatal: loose object 4ad054cb435c7091efed8e5d8ba20b3e3ab5231d (stored in .git/objects/4a/d054cb435c7091efed8

However, at the next line I can see my EUI key.

In the log file is there a relevant message I can look for?

> ln: impossible de créer le lien symbolique '/opt/ttn-gateway/bin/local_conf.json': Le fichier existe

> ### [GPS] ###
> Apr 26 18:35:50 smartsensors ttn-gateway[328]: # Invalid gps time reference (age: 1556296340 sec)
> Apr 26 18:35:50 smartsensors ttn-gateway[328]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
> Apr 26 18:35:50 smartsensors ttn-gateway[328]: ##### END #####

I also have some lines as the following

> 
> ### [UPSTREAM] ###
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # RF packets received by concentrator: 0
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # RF packets forwarded: 0 (0 bytes)
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # PUSH_DATA datagrams sent: 2 (444 bytes)
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # PUSH_DATA acknowledged: 0.00%
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: ### [DOWNSTREAM] ###
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # PULL_DATA sent: 6 (0.00% acknowledged)
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # RF packets sent to concentrator: 0 (0 bytes)
> Apr 26 18:39:20 smartsensors ttn-gateway[328]: # TX errors: 0

gonzalocasas, 2019-04-26:
``ttn-gateway[328]: fatal: loose object `` -> that looks like the git repo is broken.... maybe due to SD card corruption? I would suggest a re-install. The EUI is basically the MAC address of your ethernet interface (with some FFFE padding in between)

pierrot10, 2019-04-26:
Thank a lot for your fast answer! What do you means by 

> that looks like the git repo is broken.... maybe due to SD card corruption?

The SD card of my RPY works. I will reinstall the gateway but here is not a way to fix it?

Let me ask you a last question. When you say reinstall, do you mean by lunching again:
`sudo /home/pierrot/ic880a-gateway/install.sh`

Should I not uninstall first? Is there an option to uninstall it?
Or should I simply lunch again install.sh?

OR should I reinstall the RPY and then install the ic880a-gateway soft?

pierrot10, 2019-04-26:
And (sorry), I supposed

> If you want to use the remote configuration option, please make sure you have created a JSON file named as your gateway EUI (e.g. B827EBFFFE7B80CD.json)

as I already created a json file in your repository, I beleive, I do no need to do it again. I install following this [https://github.com/ttn-zh/ic880a-gateway](https://github.com/ttn-zh/ic880a-gateway), I restart and it's done. My EUI will not change, that correct? 

Thank a lot

gonzalocasas, 2019-04-26:
> Or should I simply lunch again install.sh?

Yes, that's enough.

> OR should I reinstall the RPY and then install the ic880a-gateway soft?

I meant a full install, strarting with a clean raspbian install, but you can try just the ic880a part first.

> My EUI will not change, that correct?

Correct, the eui will not change.

pierrot10, 2019-04-26:
Grreat, thank a lot gonzalocasas. Iam trying it right now

pierrot10, 2019-04-26:
That was not good

> error: object file .git/objects/62/002a0d748868181c53398b18c9af33c6dac128 is empty
> error: object file .git/objects/62/002a0d748868181c53398b18c9af33c6dac128 is empty
> fatal: loose object 62002a0d748868181c53398b18c9af33c6dac128 (stored in .git/objects/62/002a0d748868181c53398b18c9af33c6dac128) is corrupt
> fatal: The remote end hung up unexpectedly

I do not understand what is and where is

> .git/objects

gonzalocasas, 2019-04-26:
please delete `/home/pierrot/ic880a-gateway/` entirely and clone it again

pierrot10, 2019-04-26:
Hello, 

I confirmed that works. I just placed back the second gateway to the field and it receive the measures from the nodes [demo](http://demo.smart-idea.io/index.php). I just wonder why my first gateway could node receive the messages from the nodes while the second could, now.  Apparently, during the re-installation of the second gateway, some update have been done

```
//less ic880a-gateway/install.sh
# Update the gateway installer to the correct branch
echo "Updating installer files..."
OLD_HEAD=$(git rev-parse HEAD)
git fetch
git checkout -q $VERSION
git pull
NEW_HEAD=$(git rev-parse HEAD)

if [[ $OLD_HEAD != $NEW_HEAD ]]; then
    echo "New installer found. Restarting process..."
    exec "./install.sh" "$VERSION"
fi
```
I will reinstall the first gateway, may be it will make a difference.

Enjoy your week-end!

Erik-Koning, 2019-05-22:
Did you try again with the first gateway?

closed 2019-05-29 by pierrot10