ttn-zh/ic880a-gateway #54: Packet forwarder problem

yassinechamekh, 2019-05-21:
Hello,
I am installing a Lora WAN gateway but before I began suddenly I found that the packet forwarder has been on hold https://github.com/TheThingsNetwork/packet_forwarder  
and in the TTN-ZH I found that the install.sh will use it 
# Build packet forwarder
if [ ! -d packet_forwarder ]; then
    git clone -b legacy https://github.com/TheThingsNetwork/packet_forwarder.git
    pushd packet_forwarder
else
    pushd packet_forwarder
    git fetch origin
    git checkout legacy
    git reset --hard
fi
make
popd
so my question is that will I found a problem in sending data to TTN ? 
In the link above of packet forwarder you can see that they suggest " Semtech packet forwarder or MP packet forwarder. How to use one of them instead of packet forwarder ?
Thank you in advance.
Yassine

gonzalocasas, 2019-05-21:
@yassinechamekh this is correct; the development of the packet forwarded was put on hold, but it is still one of the projects that is most commonly used on gateways. MP is a valid alternative as well. Only recently Semtech published a new packet forwarder and we might switch to it in the future.

I will close this issue now, because there's no action to be taken, but feel free to keep commenting on it.

closed 2019-05-21 by gonzalocasas

yassinechamekh, 2019-05-21:
@gonzalocasas Thank you for your reply. So there is no harm using packet forwarded ? it will run normally ?

gonzalocasas, 2019-05-21:
Yes, it will run normally :)

yassinechamekh, 2019-05-22:
Good morning Gonzalo,
I have a question, I want to add a python code to the raspberry of gateway
which have this purpose  :
- Start counting from the first reception( first reception of data to the
gateway) and if you did not hear anything from the sensor for example in 15
min with ID : XXXXX then send an alert email .
Do you think is possible ?
I found the code of sending an email, just I want to know does the gateway
know the ID of the sensor  and is it possible ?
Thank you

gonzalocasas, 2019-05-22:
Hi! No, that's not the best way to achieve this. The payload of the packets is encrypted and you don't have the keys at the gateway level. Additionally you don't want to tie the monitoring of one sensor to one gateway, because if you move the sensor or change gateway, the sensor will still be working but your monitor will show as it if is not. 

You should instead monitor this from API/backend side. The easiest would be to setup Grafana on any server, and monitor the output of your sensor (maybe create a minimal HTTP endpoint or use a little script that gets the data from MQTT and pushes it into a database). The alerting part is already in grafana so you don't have to do anything special

yassinechamekh, 2019-05-22:
Thank you for your reply,
But I want the alert part before monitoring anything, so I only have to
make an alert at the monitoring part if I did not receive anything send me
and email or something else, is it possible on Grafana ? I did not use it
before I used cayenne myedevices.

gonzalocasas, 2019-05-22:
Yes, absolutely. Grafana has all that and it's very easy to configure. Also, alerting can be much more than just email, you can send the alerts to s different medium, e.g. slack channels.

yassinechamekh, 2019-05-22:
Thank you a lot Gonzalo, yes I find what you mean, it is easy.
One last question please, there is a way to store my data from TTN to MYSQL
?

gonzalocasas, 2019-05-22:
There are lots of options to get the data into MySQL. You can check out [this option using NodeRED](https://forum.pycom.io/topic/1034/ttn-mqtt-node-red-mysql-local-backup-of-your-lorawan-data) or [this simpler one using mostly just Javascript](https://github.com/Kaasfabriek/TTN-MQTT-To-MYSQL-AND-PHP-To-CSV), or [this other option](https://github.com/Danny-Engelman/ITpings). It all comes down to a very simple script that subscribes to MQTT and then pushes it to a database, it's literally not more than 30 lines of code in most languages.

yassinechamekh, 2019-06-06:
good morning Gonzalo,
Please I have a question.
In fact I wrote a code in the " payloads formats " of TTN exactly in the
decoder.
function Decoder(bytes, port) {
  var humidity =bytes [0]+bytes[1]/100;
    var temperature =bytes [2]+bytes[3]/100;
return {
    humidty : humidity,
    temperature : temperature
  }
}

==> it work fine all what I want to know is it coded in what language ( C,
C++, java, ...) ?
Thank you.

gonzalocasas, 2019-06-06:
It's a subset of JavaScript, executed by a Go-lang interpreter.

yassinechamekh, 2019-06-12:
Good morning Gonzalo,
So in my project I am almost done, I am coding the node in Arduino, I want
to know if the TTN library in Arduino does it support the " Dragino Lora
shield" ?
Thank you.

gonzalocasas, 2019-06-12:
No, it does not. If you go to the website, the hardware support is listed right on top and it says it only supports the Microchip RN2xxx modules: https://github.com/TheThingsNetwork/arduino-device-lib

yassinechamekh, 2019-06-12:
Thank you for your fast reply.
yes I saw that I thought that they added it in the last update, do you know
a library that support Dragino lora shield ?

gonzalocasas, 2019-06-12:
LMiC should do the trick: https://github.com/matthijskooijman/arduino-lmic/

yassinechamekh, 2019-07-01:
Hello gonzalo,
So everything work fine, I only now want to see the my data in Node RED.
I did everything like the The things network said
https://www.thethingsnetwork.org/docs/applications/nodered/quick-start.html
but it doesn't work, it always gave me error ( I put everything write ) .
Do you have any idea please ?
Thank you.

gonzalocasas, 2019-07-02:
@yassinechamekh I don't know, sorry. Best would be to post your question on the TTN forum.

yassinechamekh, 2019-07-12:
Good morning Gonzalo ,
Please I have a problem could you check it with me I can't find the problem
( the compilation is good but there is warning and in ttn I can't receive
the payload that I want ) .
You will find the Arduino code in the attachment and the warnings.
The code in Arduino you can compile if you want it will be easy for you.
Thank you in advance.

gonzalocasas, 2019-07-12:
@yassinechamekh attachments don't get attached to the github issue, so I don't see the code you mention, but on the other hand, this is not the right place to ask for help on arduino code, this repo is for the ic880a gateway installer stuff. Please, as I already mentioned, post your questions on the forum.

yassinechamekh, 2019-08-21:
Good morning GONZALO,
In fact I follow the steps that you made in GitHub for
<https://github.com/ttn-zh/ic880a-gateway#the-things-network-ic880a-based-gateway>The
Things Network: iC880a-based gateway but the gateway it disconnect a lot of
times in one day, do you have an idea ?
And one last question , I see that there is the packet forwarder in the
repository , should I follow the steps there and install it or does it
already installed when I followed the steps in ic880a gateway ( I mean in
install.sh) ?
Thank you.

gonzalocasas, 2019-08-21:
Yes, disconnects are not uncommon if you used jumper wires to connect the board, although they are not very problematic because the packet forwarder will restart automatically. But as you can see in the instructions, it's strongly recommended that you used a pcb board to interface the two, instead of the jumper wires:
[image]

In the instructions you will find the links to the different options, or you can print your own PCB if that's your thing.

Regarding your other question, the packet forwarder is installed already as part of the steps in the `install.sh`.

yassinechamekh, 2019-08-21:
Yes I am already using the backplane ( coredump) for ic880a with a
raspberry 3B+ .
But I saw that the packet forwarder has been updated 2 years ago (
<https://github.com/TheThingsNetwork/lora_gateway#v401>v4.0.1) ;
https://github.com/TheThingsNetwork/lora_gateway ( link in install.sh) but
in my case the  <https://github.com/TheThingsNetwork/lora_gateway#v301>v3.0.1
!
I tried to reinstall the install.sh , I said maybe it will update it but
nothing happen , I don't why I am stuck at that particular version and I
don't get updates.

gonzalocasas, 2019-08-27:
That's the correct version, the installer installs the `legacy` branch.