ttn-zh/ic880a-gateway #55: Typo mistake I think!

sailokeshgoud, 2019-05-22:
Please change the command while installing the packet forwarder in this [link](https://github.com/ttn-zh/ic880a-gateway)
**after these two commands: 
 $ git clone https://github.com/ttn-zh/ic880a-gateway.git ~/ic880a-gateway
 $ cd ~/ic880a-gateway**

It should be: **$ sudo ./install.sh spi**

But it is like : $ sudo ./install.sh 

**spi** is missing and it'c confusing most of the people.

gonzalocasas, 2019-05-22:
Mmm.. no, actually, the README that you're linking to is the `master` branch one, that is for the USB version, if you switch to the `spi` branch, the README correctly shows instructions with the `spi` argument in it.

closed 2019-05-22 by gonzalocasas

sailokeshgoud, 2019-05-23:
yeah thanks for the explanation! I got it now!