ttn-zh/ic880a-gateway #56: Strange behavior from my gateway

pierrot10, 2019-05-29:
Hello All,
I have a strange behavior and I would like to know if you can help to understand.

Let me first describe how is my structure.

I have 2 nodes in a orchard at about 450m altitude. There are not above the tree and the foliage, but on the top (you we will better understand later). One of station is in the middle of the orchard and the second at the end.

The gateway it on the top of my camping car, at about 900m of the first station and about 1100m of the second.

I know that my gateway is not well located because it is at a lower altitude than my nodes and there are winyard between them.

Between June and September 2018, in general the transmission were fine.

End of April 2019, I turned on my nodes and the gateway for new measurements, while the trees did not have foliage yet and the transmission started well (the nodes transmit each hour). The leafs appeared and grew up. And suddenly, at the same time (within one hour) , the gateway do not receive and transmit the messages from the nodes.

**As it happen at the same time**, I though about a problem with the gateway, but not. Then I moved the gateway at about 500m from the nodes, almost at the same altitude and the area is clearer. The nodes stated to transmit again. Great!

It ran well for about 2 weeks until my Gateay MicroCard broke. I changed and reinstalled it. The nodes transmitted again but only fro 4 hours.

** nodes stop transmitting at the same time again**. Or let say may be , the gateway do not receive the message, because the nodes has not stoppred working. For me, its still transmit.

I thought again an issue with the new installed gateway, but not, the gateway worked (I can see the green status at TTN console . Then, I moved the gateway at about 50m from the nodes and the nodes started transmitting!

My guess is as the foliage “covers” the nodes (even if there are at the top of the tree), the foliage act as an obstacle, but it can not explain why the two nodes stopped transmitting at the same time, first all because the apple tree has very less foliage at the second node. My guess is the foliage would make an obstacle, the node will not stop transmitting during the same hour, but may be, with a couple of days of delay.

Then I would to understand why this behavior?
Why the node stop transmitting at the same time, and why it works when I move closer the gateway, while last year, it worked at 1km of distance.

Can it be the power of transmitting, how changed automatically?
**Can it be possible that the gateway’s range (sensitivity) somehow dropped and how can I check it and fix it?**

I configured as in the example: https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-abp/ttn-abp.ino#L218

Note, I am using for now (but I will change to OTAA) ABP activation method.

Thank for your point of view and have a nice evening/day.
Cheers

pierrot10, 2019-06-02:
I hope you can help me soon. I oberserved the problem come since I reinstall both gateway with the last rasbian. None of them work perfectly (almost nerver) . 
But it's really suprising. Today I psent my time in the field with my two gateway. On was working. I built. a node to measure the RSSI and I could work aroud the field from each point. Then as that gateway was working, I moved the hardware into a bigger box (which is the same as the second, but a bit bigged). I switch on and not working any more. Then I used the second gateway not working and I give up and I left the gateway in the field

I am just in my flat, I plug and switch on the gateway I bring back with me, .... and it works!!!!

My nodes are in a orchard and now there are leafs around the nodes, but my node 2 is at 10m from my gateway and last year it was working well, while my gateway was at 1km from my two nodes.

So what can I do now to make sure that my gateway work at any time, ... and any where?

Many thank

gonzalocasas, 2019-06-02:
There are too many things that could be going wrong, it's not something that can easily be diagnosed over comments, but here is an unordered list of things to consider:
- First of all, please describe exactly your gateway setup. I assume it's an ic880a, but please describe your enclosure, board to board connector, power supply, and antenna.
- Second, location is everything when it comes to gateways, and you start off saying "I  know that my gateway is not well located", so that's what you should be focused on first.
- Foliage can be a problem but is unlikely to affect two nodes at the same time, but perhaps, if your nodes are at the edge of what works, a little interference from foliage makes everything fail. Reasons for this could be, for example: a bad quality antenna, incorrect spreading factor/power settings on the nodes (you mention you still use ABP so there's no auto management of SF taking place I assume). Check the meta data of the messages that do work and then decide whether that has anything to do.
- Locations: you mention your nodes are at 450m altitude, is that above sea level, I assume? What's the ground altitude? And your gateway, is it really on top of your camping car?? That won't get you far, on those low altitude installations I have never seen more than some hundred of meters, max 700m but not much more. This could really be the root of the problem.
- Last, be methodic about diagnosing this. And add as much info as possible here (logs, photos, details, etc) if you expect to get some help, otherwise, as mentioned, is really hard to even start.

Cheers!!
Gonzalo

pierrot10, 2019-06-02:
Dear gonzalocasas,
Thank a LOT for your reply.

> First of all, please describe exactly your gateway setup. I assume it's an ic880a, but please describe your enclosure, board to board connector, power supply, and antenna

I followed that [exactely procedure](https://github.com/ttn-zh/ic880a-gateway). 
The hardware are:
[Concentrator](https://shop.imst.de/wireless-modules/lora-products/8/ic880a-spi-lorawan-concentrator-868-mhz)
Raspberry Pi2 ou 3
[A baclkpane](https://www.tindie.com/products/gnz/imst-ic880a-lorawan-backplane-kit/)
[Antenna](https://shop.imst.de/wireless-modules/accessories/19/sma-antenna-for-ic880a-spi-wsa01-im880b-and-lite-gateway)
Into the orchard, the gateway is powered by a Gel 12V/100Ah with a 65W solar panel. Tomorrow, I will go into the orchard to take pictures. But as I wrote, **last year it worked very fine** with that configuration.
In my flat, the gateway is the same, same hardware, same box (a bit smaller, but same box) and powered by the wall plug.
[photo 1](http://www.smart-idea.io/temp/gtw-home/1.JPG)
[photo 2](http://www.smart-idea.io/temp/gtw-home/2.JPG)
[photo 3](http://www.smart-idea.io/temp/gtw-home/3.JPG)
[photo 4](http://www.smart-idea.io/temp/gtw-home/4.JPG)
[photo 5](http://www.smart-idea.io/temp/gtw-home/5.JPG)
[photo 6](http://www.smart-idea.io/temp/gtw-home/6.JPG)

> **Second, location is everything when it comes to gateways, and you start off saying "I know that my gateway is not well located", so that's what you should be focused on first.**

Yes, that was when it was on the top of my camping car at 1km from the nodes. Last year it worked.
But now and the previous days, I moved the gate at 500m, the 200m at the top of a small hill with no obstacle, and now the gateway is at 10m and may 80m form my node. So at at the moment, there are well located excepted the foliage ([see picture of one of the node](see picture of one of the node)). But why last year, it have worked? Tomorrow, I will move the gateway at the same "axe", but again, why it worked last year, and why it worked 2 week and why **the measure of the two nodes was not receiving at the same time**. This really suprise me. If the foliage would be the problem, the gateway will not missing the measure of the node at the same time!!?? Are you agree?

> Foliage can be a problem but is unlikely to affect two nodes at the same time, but perhaps,

Yes, it was my first thougth, but since I oberserved the gateway missed the measure of the two node at the same time, I supected the gateway. Additonally, today it work the time I worked around the orchard and until I switch off the gateway. I restart, and .... trouble..

> if your nodes are at the edge of what works, a little interference from foliage makes everything fail. 

I am agree, but again, why last year I did not experiment that issue and we are at the begining of the season. The second node is at 10m from my gateway and we can see it. Do you think that I 10m I am at the edge of what work?

> incorrect spreading factor/power settings on the nodes 

I use the default as you can [see here](https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-abp/ttn-abp.ino#L218)

**How can I check my gateway regarding this topics?** Your code is by default DR_SF7,14

>(you mention you still use ABP so there's no auto management of SF taking place I assume). 

Yes, that's correct, with no auto management of SF. (I do not remember how to change to auto)

> Check the meta data of the messages that do work and then decide whether that has anything to do
Tomorrow, I will retrun to the orchar with TTN console open to see if I can read what you suggest.

> you mention your nodes are at 450m altitude, is that above sea level, I assume? What's the ground altitude? And your gateway, is it really on top of your camping car?? That won't get you far, on [...]

Yes more or less. the roof of my camping car is 3.2m from the ground and the camping car is at about 1km. But thta work for one year, even if the gateway location is not perfect. Now the gateway is at 10m and does not work. As I wrote, I also place the gateway at about 200,, 500m without success while it works at 1km, last year.....

> Last, be methodic about diagnosing this. And add as much info as possible here (logs, photos, details, etc) if you expect to get some help, otherwise, as mentioned, is really hard to even start.

Yes, I understand, I will ad more picture of the station.
For now I can show you [the map](http://smart-idea.io/measures/map_montfort.php) of the gateway and sensor
Yellow are the nodes
**Violet is where is now the gateway (zoom to see it)**
Blue is the first location of the gateway (roof of my camping car and work from spring until october 2019)
Pink is the second location of the gateway (it has been working for some day
Red is the third location (it never work), before I move the gateway to the violet location

Tomorrow, I will take picture of the foliage

Hope my comment can help, you can help me to clarify that issue

Many thank

.

pierrot10, 2019-06-02:
I forgetten to write simething importa, I wrote I created a node to measure the RSSI which send message each 30 seconde. Even when I was sitting beside my gateway, the gateway do not transmit the measure of the RSSI node. It's the reason why, I do not think the foliage is the major obstacle.

pierrot10, 2019-06-06:
Hello, Tuesday, I reinstalled my gateway with Jessy. It has been working for four hours, then nothing for 3 hours. Nothing for about 24h and a mark for today at 13:38
[Station 1](http://demo.smart-idea.io/index.php?st=1)

Here are the pcitures, I took on Wednesday
At this time the gateway was at [Violet position](http://smart-idea.io/measures/map_montfort.php)
[Gateway](http://www.smart-idea.io/temp/gateway.JPG)
[Gateway](http://www.smart-idea.io/temp/gateway1.JPG)
[Gateway seen from the station 2](http://www.smart-idea.io/temp/gateway-from-st2.JPG)
[Station 2 (at 10-20m)](http://www.smart-idea.io/temp/st2.JPG)
[Station 1 (at 100-150m)](http://www.smart-idea.io/temp/st1.JPG)
From the 4th until today the 6th, the transmistion are better from Station 1 than Station 2
[Station 1](http://demo.smart-idea.io/index.php?st=1)

gonzalocasas, 2019-06-06:
Most of the images you linked do not work/return a 404-like page: 
[image]

gonzalocasas, 2019-06-06:
Also, the maps you posted contain only the gateway in violet, but no nodes are visible there: 
[image]

gonzalocasas, 2019-06-06:
I wouldn't know what the problem is; it can be almost anything, the failures don't have a pattern. It could be temperature (did it fail during night's lower temperatures?) affecting the modules (the RN2483 first rev are known to fail on lower temps); it could be assembly problems (the antenna you showed on your indoor box has a clearly indoor antenna, you mention your gateway is more or less the same, but I hope you have an outdoor antenna instead), it could be power supply issues on the gateway side, it could be code issues, you mention you used "the default SF7", but that's not the default, that's the optimal SF, you should be using OTAA and AdR for proper SF management.

For now, the only thing I could say is, check all these points one by one, make sure every piece of the puzzle works in isolation.

pierrot10, 2019-06-06:
Dear gonzalovasass,
I am so sorry, I missed a directory. The image works now.
Regarding the map, I just tried and I can see [all marks](http://smart-idea.io/measures/map_montfort.php). The nodes are the yellow mark. Blue, pink and red, was the previous location of the gateway
(I am reading your last comment)

pierrot10, 2019-06-06:
Dear Gonzalocasas,
The lowest temperature from the 3rd of June was 12°C. For the station 1 , it stopped at 3.34 UTC, and the last was at 13:38 UTC, next day. The station 2 stopped at 3:38 UTC as well and the gateway never get new measure. The nodes have the same antenna than the gateway.

If you look at the graph, and change the date from the 1st of may, we can see it work well until the 20th of may and the temperature was never below 5°C. On the 30th of may, the gateway get 4 measures and stop at 13:53UTC and the temperature was 23°C.

Yes my two gateway and nodes have the same antenna and I did not know, there were indoor andtennas!!!!????

pierrot10, 2019-06-06:
> but that's not the default, that's the optimal SF

But a gateway is fixed on a special SF? I mean, do I have to set my node SF regarding the Gateway? If yes, how can I modify the gateway SF, or what's the best SF of my node, for the gateway?

gonzalocasas, 2019-09-23:
@pierrot10 sorry for the very long delay. Regarding the SF questions, gateways don't have a fixed SF, they will be listening for all of them. It's your node that needs to regulate that, if you're having reach problems, just bump the SF7 to SF9 for instance. But ideally, as mentioned, you should be using Adaptive Data Rate (ADR) so that the data rate is managed automatically by the network. 

I don't think the problems you mention are related to the source code in this repository, so I'm going to close this issue; but feel free to keep commenting anyway.

closed 2019-09-23 by gonzalocasas