ttn-zh/ic880a-gateway #60: gateway no longer visible

thl-mot, 2019-08-18:
since 20hours my gateway is not seen by ttn after working for 2 years now without any known errors.
Looking at syslog ist seams to work like always. no errors. 
Where should I look for the error.
thomas

thl-mot, 2019-08-18:
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### 2019-08-18 19:13:05 GMT #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [UPSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets received by concentrator: 0
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets forwarded: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA datagrams sent: 1 (223 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA acknowledged: 100.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [DOWNSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_DATA sent: 3 (100.00% acknowledged)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # TX errors: 0
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [GPS] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Invalid gps time reference (age: 1566155585 sec)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Manual GPS coordinates: latitude 49.03659, longitude 8.44918, altitude 0 m
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### END #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 38 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 36 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 38 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 37 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### 2019-08-18 19:13:35 GMT #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [UPSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets received by concentrator: 1
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets forwarded: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA datagrams sent: 1 (223 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA acknowledged: 100.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [DOWNSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_DATA sent: 3 (100.00% acknowledged)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # TX errors: 0
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [GPS] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Invalid gps time reference (age: 1566155615 sec)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Manual GPS coordinates: latitude 49.03659, longitude 8.44918, altitude 0 m
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### END #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 40 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 37 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 37 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 38 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### 2019-08-18 19:14:05 GMT #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [UPSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets received by concentrator: 0
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets forwarded: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA datagrams sent: 1 (223 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PUSH_DATA acknowledged: 100.00%
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [DOWNSTREAM] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_DATA sent: 3 (100.00% acknowledged)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # TX errors: 0
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ### [GPS] ###
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Invalid gps time reference (age: 1566155645 sec)
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: # Manual GPS coordinates: latitude 49.03659, longitude 8.44918, altitude 0 m
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: ##### END #####
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 41 ms
Aug 18 21:14:21 ttn-gateway ttn-gateway[397]: INFO: [down] for server router.eu.thethings.network PULL_ACK received in 37 ms

gonzalocasas, 2019-08-18:
The logfile you post looks totally normal. According to [TTN's status page](https://status.thethings.network/), routing services of the EU region are experiencing performance issues. Perhaps that's the cause of the problems you are seeing.

I would recommend posting this on the global Slack (#ops channel) or the forum, or perhaps, just to test, try switching to a different router, like the swiss one, or some other, just to see if that works better.

In any case, I will close this issue, because it's really unlikely this is a code issue related to this repository.

closed 2019-08-18 by gonzalocasas

kizniche, 2019-09-09:
I also have my gateway showing "not connected" in TTN, however data is still being received my all my nodes. It has been showing as "not connected" for 22 days as of today, which is exactly around the time @thl-mot started this issue. I have another gateway that doesn't have this issue. There don't seem to be any errors in the log.

[image]

[image]

The router these gateways are connected to appears to be working fine.

[image]

Can you verify other gateways using this code is working as expected?

kizniche, 2019-09-09:
I'm also discussing the issue in The Things Network Slack #ops channel to see if they can diagnose any issue on their end.

gonzalocasas, 2019-09-09:
@kizniche 👍  please post here once you have a definitive answer.

kizniche, 2019-09-10:
Changing the server in the config file from "router.us.thethings.network" to "router.au.thethings.network" caused the router status in TTN console as "connected". A user on slack said this may interfere with OTAA downlink frequencies, however I'm currently only using ABP, so I can't immediately test this.