ttn-zh/ic880a-gateway #61: Problem using remote settings file

CRosemaryC, 2019-08-27:
Hello, thanks for creating the push, but do you know why I am having this error now?

error: The following untracked working tree files would be overwritten by merge:
        B827EBFFFEFCDAA2.json
Please move or remove them before you merge.
Aborting

Thank you.

CRosemaryC, 2019-08-27:
> You probably did the change locally before pushing it. You can either 1)
> remove all that and reinstall 2) or do a git reset --hard HEAD and try
> again.
>
> I'm closing this issue because there's no code change needed on this repo,
> but feel free to keep commenting if the suggested fix doesn't work for you.

Thank you very much for the information.

I am new in this, and when I tried the 2nd option, the output is:
HEAD is now at aa092b0 Update start.sh to match interface name detection on
install
-bash: HEAD: command not found
Do you know how to update start.sh?

Also I would know how to remove everything, is enough to just delete the
git clone folder?

Best regards,

Cristina.

CRosemaryC, 2019-08-27:
Never mind, I manage to make it work.

Best regards,

Cristina.

CRosemaryC, 2019-08-27:
> Great to hear that! 👍

I still have no connection status in ttn console.

Is it normal that while running the install.sh, there some pink and blue
text.
e.g.
src/poly_pkt_fwd.c:694:3: warning: ‘strncpy’ specified bound 64 equals
destination size [-Wst
 ringop-truncation]
strncpy(monitor_addr, str, sizeof monitor_addr);

 Also, this works for all types of LoRa chipsets right? Should I change in
the start.sh the reset pin number if mine is different than yours?

Best regards,

Cristina.

gonzalocasas, 2019-08-27:
You probably did the change locally before pushing it. You can either 1) remove all that and reinstall 2) or do a `git reset --hard HEAD` and try again.

I'm closing this issue because there's no code change needed on this repo, but feel free to keep commenting if the suggested fix doesn't work for you.

closed 2019-08-27 by gonzalocasas

gonzalocasas, 2019-08-27:
Great to hear that! 👍

gonzalocasas, 2019-09-30:
> Also, this works for all types of LoRa chipsets right?

Not really, it works for ic880a boards. Any other board might or might not work, but we haven't tested them all.

> Should I change in the start.sh the reset pin number if mine is different than yours?

Yes, of course. If the board is based on the same Semtech chipset, over the same SPI interface, but only different reset pin, that will probably work. If it's another chipset, no idea.