ttn-zh/ic880a-gateway #62: Question about updating

dr-eamy, 2019-09-29:
Hi, Installed this git on my raspberrypi 0 w whit iC880A and iC880A LoRaWAN Gateway Backplane v2.0  some time ago, and i guess i need to update it from time to time... are there any place somebody has written a how to ?

E

gonzalocasas, 2019-09-29:
Just run the installer again:
```
  $ cd ~/ic880a-gateway
  $ sudo ./install.sh spi
```

closed 2019-09-29 by gonzalocasas

SchneHa, 2021-11-05:
Hello,

sorry for opening this issue again.

I tried to update to migrate my RPi/iC880a Gateway to V3 and I got the following error message:

ttn@ttn-gateway:~/ic880a-gateway $ sudo ./install.sh spi
The Things Network Gateway installer
Version spi
Updating installer files...
error: Ihre lokalen Änderungen in den folgenden Dateien würden beim Auschecken
überschrieben werden:
	src/sensors_js/install.sh
	src/sensors_js/sensors-js.service
	src/sensors_js/sensors.js
Bitte committen oder stashen Sie Ihre Änderungen, bevor Sie Branches
wechseln.
Abbruch
Indeed, I did changes to the files to use some sensors and wouldn't like to loose it. What can I do?

BR,
Hans