ttn-zh/ic880a-gateway #63: error: pathspec 'legacy' did not match any file(s) known to git.

1012207826, 2019-10-03:
I meet a issue that"error: pathspec 'legacy' did not match any file(s) known to git.",and I don't now how to solve it.

gonzalocasas, 2019-10-03:
I would need a lot more information to understand what might be wrong. When exactly do you get this error? Which setup do you have? Anything different from the instructions on the wiki?

1012207826, 2019-10-04:
Firstly,Thank you give me replay，and I am very happy.
I want to install loraserver ,and I have downloaded files from github.But I met a promlbe when I was Installing the server。following the pictures.

then I use the commands:pi@rak-gateway:~/ic880a-gateway $ sudo ./install.sh ips

From the above picture ,I chose the install it from local file,so I met the follow problem,and I don't know where I should configure the file of json.
Can you help me? Thank you

gonzalocasas, 2019-10-04:
`sudo ./install.sh ips` -> that is wrong. The last argument should be `spi` not `ips`.

closed 2019-11-06 by gonzalocasas