ttn-zh/ic880a-gateway #66: "not connected" on TTN.

shavaizknz98, 2020-02-08:
Hello, I am currently creating a LoRaWAN outdoor multi-channel gateway fora university project. I purchased a RPi3 B+ and an ic880A and followed the guide on https://github.com/ttn-zh/ic880a-gateway/wiki, I even added my gateway ID to the remote-config repo. However, by the end of the installation my gateway status is “not connected” on TTN. Can anyone please help out. Here are some logs from my RPi.

After running sudo systemctl status ttn-gateway.service I receive the following,

```
ttn@ttn-gateway:~ $ sudo systemctl status ttn-gateway.service
● ttn-gateway.service - The Things Network Gateway
Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
Active: active (running) since Sat 2020-02-08 12:42:02 +04; 11min ago
Main PID: 317 (start.sh)
Tasks: 4 (limit: 2200)
Memory: 16.2M
CGroup: /system.slice/ttn-gateway.service
├─317 /bin/bash /opt/ttn-gateway/bin/start.sh
└─686 ./poly_pkt_fwd

Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: ##### 2020-02-08 08:52:55 GMT #####
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: ### [UPSTREAM] ###
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # RF packets received by concentrator: 0
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # RF packets forwarded: 0 (0 bytes)
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # PUSH_DATA datagrams sent: 1 (287 bytes)
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # PUSH_DATA acknowledged: 0.00%
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: ### [DOWNSTREAM] ###
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # PULL_DATA sent: 3 (100.00% acknowledged)
Feb 08 12:52:55 ttn-gateway ttn-gateway[317]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)

```
And after running sudo tcpdump -AUq port 1700 I get,

```
ttn@ttn-gateway:~ $ sudo tcpdump -AUq port 1700
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
12:57:54.091133 IP ttn-gateway.52747 > 52.169.76.203.1700: UDP, length 12
E…(u.@.@.W…+.4.L…B$…s…’…cJ
12:57:54.344801 IP 52.169.76.203.1700 > ttn-gateway.52747: UDP, length 4
E . …@.-…4.L…+…Hd…s.
12:57:55.962624 IP ttn-gateway.54597 > 52.169.76.203.1700: UDP, length 287
E…;u.@.@.U…+.4.L…E…‘oh.;…’…cJ{“stat”:{“time”:“2020-02-08 08:57:55 GMT”,“lati”:25.31234,“long”:55.49102,“alti”:0,“rxnb”:0,“rxok”:0,“rxfw”:0,“ackr”:0.0,“dwnb”:0,“txnb”:0,“pfrm”:“IMST + Rpi",“mail”:"Shavaizknz98@gmail.com”,“desc”:“LoRaWAN gateway for Smart Campus Mobility System at the American.’…cJ.”}}
12:58:04.221134 IP ttn-gateway.52747 > 52.169.76.203.1700: UDP, length 12
E…(xG@.@.Tu…+.4.L…|…’…cJ
12:58:04.538464 IP 52.169.76.203.1700 > ttn-gateway.52747: UDP, length 4
E . …@.-…4.L…+…

```
After running sudo tail -f /var/log/syslog , I get:

```
ttn@ttn-gateway:~ $ sudo tail -f /var/log/syslog
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # RF packets forwarded: 0 (0 bytes)
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # PUSH_DATA datagrams sent: 1 (287 bytes)
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # PUSH_DATA acknowledged: 0.00%
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: ### [DOWNSTREAM] ###
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # PULL_DATA sent: 3 (0.00% acknowledged)
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # RF packets sent to concentrator: 0 (0 bytes)
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # TX errors: 0
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: ### [GPS] ###
Feb 8 13:00:56 ttn-gateway ttn-gateway[317]: # Invalid gps time reference (age: 1581152455 sec)
```

Any and all help will be appreciated :)

gonzalocasas, 2020-02-08:
What's your gateway EUI?

shavaizknz98, 2020-02-08:
Do you mean the gateway ID? Here it is B827EBFFFEBF634A

tmallikarachchi, 2020-05-01:
Did you find a solution for this?
I'm running into a similar issue. In my case, I get a mixed from time to time. Here the are the outputs when I run `sudo systemctl status ttn-gateway.service`

ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-05-01 18:06:16 BST; 329ms ago
 Main PID: 2241 (start.sh)
    Tasks: 2 (limit: 2200)
   Memory: 512.0K
   CGroup: /system.slice/ttn-gateway.service
           ├─2241 /bin/bash /opt/ttn-gateway/bin/start.sh
           └─2253 sleep 0.1

May 01 18:06:16 ttn-gateway systemd[1]: Started The Things Network Gateway.
pi@ttn-gateway:~ $ ./checkstatus.sh 
ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2020-05-01 18:06:22 BST; 4s ago
  Process: 2272 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
 Main PID: 2272 (code=exited, status=1/FAILURE)

May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Radiostream data is enabled
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Statusstream data is enabled
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Beacon is disabled
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Monitor is disabled
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Contact email configured to ""
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Description configured to "ttn-ic880a"
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: Successfully contacted server router.eu.thethings.network
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: INFO: [main] Starting the concentrator
May 01 18:06:22 ttn-gateway ttn-gateway[2272]: ERROR: [main] failed to start the concentrator
May 01 18:06:22 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.

I'm using RPi 3. The status in things network is always seen as "not connected."

shavaizknz98, 2020-05-01:
My issue was fixed a couple of days after. I just reset the RPi and followed all the instructions again. Magically worked somehow, still do not know what the original issue was.

closed 2020-05-01 by shavaizknz98

tmallikarachchi, 2020-05-01:
Thats great. Could you please be able to confirm whether the pin configuration in the description matches the diagram? My setup seem a bit different compared to the picture in the wiki page.

gonzalocasas, 2020-05-03:
@shavaizknz98 thanks for updating.
@tmallikarachchi these issues are always related to some deviation of the instructions, or some misconfiguration. Please double and triple check your wiring is the same as in the instructions and you followed all steps correctly.

tmallikarachchi, 2020-05-03:
I managed to resolve this yesterday!. It was an issue with pin configuration in raspberry pi. I had connected the pins in RPi incorrectly. Followed the layout given in
https://www.raspberrypi.org/documentation/usage/gpio/