ttn-zh/ic880a-gateway #68: Failed to start the concentrator

thanujarc, 2020-05-01:
I'm following the wiki page to setup the LoRa Gateway using RPi3. I followed the steps managed to register the gateway in thingsnetwork. However, the status of the gateway is displayed as "not connected".

I get the following output when I run, `sudo systemctl status ttn-gateway.service`.

ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2020-05-01 19:41:24 BST; 891ms ago
  Process: 13264 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
 Main PID: 13264 (code=exited, status=1/FAILURE)

May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Radiostream data is enabled
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Statusstream data is enabled
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Beacon is disabled
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Monitor is disabled
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Contact email configured to ""
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Description configured to "ttn-ic880a"
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: Successfully contacted server router.eu.thethings.network
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: INFO: [main] Starting the concentrator
May 01 19:41:24 ttn-gateway ttn-gateway[13264]: ERROR: [main] failed to start the concentrator
May 01 19:41:24 ttn-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.

I also noticed that my final pin layout after connecting the ic880a board to RPi3 is different to that of the picture in the wiki page. 
I have attached an image of my pin configuration in the link below. 
https://1drv.ms/u/s!Ak9WeKMxuttok9dijCJgaJHz2gKPkg?e=AfAgXf

Can someone please help me to sort out this issue?

thanujarc, 2020-05-01:
This is now resolved. My pin configuration was incorrect. I finally used the correct GPIO layout and it solved the problem. 
https://www.raspberrypi.org/documentation/usage/gpio/

closed 2020-05-01 by thanujarc

gonzalocasas, 2020-05-03:
Thanks for updating.