ttn-zh/ic880a-gateway #69: How can I use teh IC880 gateway without remote configuration

pierrot10, 2020-05-26:
Hello
For some reason, my gateway does not wokrs since I switch it off.
I installed it with remote configuration option.
I wonder how I could it make it working with TTN, if I do not use that option.
Should I enter somewhere the gateway ID from my TTN console?

Cheers

gonzalocasas, 2020-05-27:
The Gateway ID is determined automatically based on your network interfaces (it's derived from your MAC address). If you don't want to fiddle to much, the easiest option is to re-install the package forwarder and answer `No` to the question of using remote configuration. It should overwrite the existing settings.

pierrot10, 2020-06-01:
Hello, thank for your reply. I did it but I still have a problem. The gateway is not seen from TTN console while it was, until I moved the gateway to another location. Now I put back to is original location (ma flat) and no way. It not seen from TTN console. The only task I did is to update the stretch OS (sudo apt update && sudo apt upgrade)

pierrot10, 2020-06-01:
Oups sorry, it's now connected !!!!!!!!!!!! Amazing!! How can it be possible. I did not update the OS today but some days ago. Today I just switch it on, I reinstalled it by anserwring "no" to use the remote configuration. I did nothing else more

closed 2020-11-06 by gonzalocasas