ttn-zh/ic880a-gateway #71: Raspberry 4

tobiascrackz, 2020-10-23:
Did someone try to make this project work with a Raspberry 4? If so, what changes in wiring or other things to consider exist to make the gateway run?

DefProc, 2020-10-23:
The RPi 4 is hardware compatible with the others, so I'd expect it to work just fine. The gateway doesn't need the extra power though, they spend most of their time idle so there's no benefit to using a Pi4 if it's just forwarding LoRaWAN packets.

tobiascrackz, 2020-10-23:
@DefProc thanks for your quick answer 👍 
> spend most of their time idle

that's the reason why other services can be run on the same raspi (MQTT, Blynk etc.) and profit from extra capacity/resources

would be nice to know if someone is sucessfully running the gateway on RasPi4.

0x0c00d843, 2020-12-14:
Hi,
I just configured a gateway on raspberry pi 4 and it works very well. The wiring is the same as for othr raspberry versions