ttn-zh/ic880a-gateway #74: No connected

baptgauthier, 2020-12-22:
Same issue on : https://github.com/ttn-zh/ic880a-gateway/issues/66

On https://console.thethingsnetwork.org/ it is indicated not connected
I redone the procedure 2 additional times but it has not changed

`ttn@ttn-gateway:~ $ sudo systemctl status ttn-gateway.service
● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-12-22 08:12:34 CET; 4h 51min ago
 Main PID: 352 (start.sh)
    Tasks: 4 (limit: 2063)
   CGroup: /system.slice/ttn-gateway.service
           ├─352 /bin/bash /opt/ttn-gateway/bin/start.sh
           └─562 ./poly_pkt_fwd

Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # PUSH_DATA acknowledged: 0.00%
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: ### [DOWNSTREAM] ###
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # PULL_DATA sent: 3 (0.00% acknowledged)
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # RF packets sent to concentrator: 0 (0 bytes)
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # TX errors: 0
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: ### [GPS] ###
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # Invalid gps time reference (age: 1608638488 sec)
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: # Manual GPS coordinates: latitude 10.00000, longitude 20.00000, altitude -1 m
Dec 22 13:01:58 ttn-gateway ttn-gateway[352]: ##### END #####`

I don't know what to do to debug and find the problem.
Could you help me ?
thanks !!

gonzalocasas, 2020-12-26:
That big of logfiles seems to indicate the concentrator is working correctly and it is the console that is not updating the status of the gateway correctly. This happens sometimes (unfortunately), so it cannot really be relied upon. For a better test, please use a lorawan module/sensor and send some packets, and check if you see them (either in the logs or in TTN).

baptgauthier, 2020-12-28:
Thank you for your reply. I will try to connect a sensor.
Where is the log file that you indicate to me stored?
Thanks !

gonzalocasas, 2021-01-04:
I mean, the systemd log that you see in your own snippet. I think you can find it on `/var/log/syslog` and also using `journalctl` for the `ttn-gateway` unit.