ttn-zh/ic880a-gateway #76: TTN V3 Migration

tobiascrackz, 2021-03-03:
Hello all :)
Is there already a guide on how to migrate this Gateway (Raspi + IC880) to TTN V3?

Gepro-TeusVerbeek, 2021-03-10:
I just tried it. It seems very simple.

I have a local configuration so I just updated local_conf.json in /opt/ttn-gateway/bin. I changed the server address from **router.eu.thethings.network** to **eu1.cloud.thethings.network**. 

If you have remote configuration configured you can change it in your remote file.

I think the script needs an update where the correct server address is selected whether you want to connect with stack V2 or V3. Or maybe the server address is automatically updated at some point so that the V3 stack gets the **router.eu.thethings.network** address.

gonzalocasas, 2021-03-10:
Yes, the gateway migration is very simple but please do not do it generally just now. If you check the forum, or ttnmapper and many other resources, you'll find that applications need to be migrated first, otherwise, devices not compatible with v3 will stop being able to send data. So, we're actually postponing the gateway migration for a later time in the year.

dbrgn, 2021-05-25:
There's a note now in the docs:

> Since version 3.13 (released in May, 2021), The Things Network V2 routes traffic back and forth to The Things Stack Community Edition. When migrating your gateways to The Things Stack Community Edition, the coverage of the public community network won’t be impacted. 

This means that we can start migrating our gateways now, right?

gonzalocasas, 2021-05-27:
Oh cool! Yep, that might indicate so! Thanks for pointing it out!

MrSimonEmms, 2021-05-28:
Does anyone have any working settings for the new v3 console? As far as I can see, there's no "legacy packet forwarder" option on the new version. I've pressed lots of buttons, but I can't seem to get it connecting

hatyri, 2021-06-19:
Is there an address specification somewhere else than Install.sh on line 182. If not, is it enough?

change
    echo -e "{\n\t\"gateway_conf\": {\n\t\t\"gateway_ID\": \"$GATEWAY_EUI\",\n\t\t\"servers\": [ { \"server_address\": \"router.eu.thethings.network\", \"serv_port_up\": 1700, \"serv_port_down\": 1700, \"serv_enabled\": true } ],\n\t\t\"ref_latitude\": $GATEWAY_LAT,\n\t\t\"ref_longitude\": $GATEWAY_LON,\n\t\t\"ref_altitude\": $GATEWAY_ALT,\n\t\t\"contact_email\": \"$GATEWAY_EMAIL\",\n\t\t\"description\": \"$GATEWAY_NAME\" \n\t}\n}" >$LOCAL_CONFIG_FILE

to

    echo -e "{\n\t\"gateway_conf\": {\n\t\t\"gateway_ID\": \"$GATEWAY_EUI\",\n\t\t\"servers\": [ { \"server_address\": \"eu1.cloud.thethings.network\", \"serv_port_up\": 1700, \"serv_port_down\": 1700, \"serv_enabled\": true } ],\n\t\t\"ref_latitude\": $GATEWAY_LAT,\n\t\t\"ref_longitude\": $GATEWAY_LON,\n\t\t\"ref_altitude\": $GATEWAY_ALT,\n\t\t\"contact_email\": \"$GATEWAY_EMAIL\",\n\t\t\"description\": \"$GATEWAY_NAME\" \n\t}\n}" >$LOCAL_CONFIG_FILE

colonelmoutarde68, 2021-07-02:
I tried to upgrade my gateway to TTN v3 by updating the server address to eu1.cloud.thethings.network, and then by recreating it (with a new name) and event reinstalling the whole thing, but no success, it always says "• Disconnected" !

umbertowski, 2021-07-06:
> I tried to upgrade my gateway to TTN v3 by updating the server address to eu1.cloud.thethings.network, and then by recreating it (with a new name) and event reinstalling the whole thing, but no success, it always says "• Disconnected" !

i have the same problem

gonzalocasas, 2021-07-19:
the script is now updated to point to v3 by default. please try and let me know. Later this year, we'll do a bigger update and switch to basicstation.

Gerard880, 2021-07-21:
Hello Gonzalocasas,
Where can i find the updated script? 
regards,
Gerard.

TTNIdrija, 2021-08-01:
$ cd ~/ic880a-gateway
$ sudo ./install.sh

dbrgn, 2021-08-01:
Full migration guide if you use local configuration:

- Create gateway in the new backend (https://console.cloud.thethings.network/)
- On the Raspberry Pi, use "git pull" to fetch the latest version of the setup script
- Run `sudo ./install.sh spi` (or without "spi" if you have the non-SPI version)
- Ensure that the gateway shows up as connected in the new backend
- Remove gateway from old backend

If you use remote configuration, then I think you'll simply have to update your config in https://github.com/ttn-zh/gateway-remote-config instead, and point it to `eu1.cloud.thethings.network`, followed by `sudo ./install.sh spi`.

colonelmoutarde68, 2021-08-01:
Tried to reinstall the whole thing with the new script but still no luck (i880a spi version)

MystikReasons, 2021-08-29:
I have the exact same problem. The 880a is working just fine with the integrated test scripts on the RPI 4 but I can't get it connected to the TTN network.

dechantw, 2021-08-29:
Has somewone an idea where i can found the log files on the gateway. I m not so familar with Linux.

dbrgn, 2021-08-29:
You should be able to follow the log in real-time with `sudo journalctl -u ttn-gateway -f`. Use `sudo journalctl -u ttn-gateway -r` to view the logfile in reverse order. Or `sudo journalctl -u ttn-gateway --since="1h ago"` to view the log of the last hour.

dechantw, 2021-08-29:
Thats the log, I don t know what it means. On the other side I have tried to add a sensor under v3, its close to my gateway, but also no success.
ttn@ttn-gateway:~/ic880a-gateway $ sudo journalctl -u ttn-gateway -f
-- Logs begin at Thu 2016-11-03 18:16:42 CET. --
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # RF packets forwarded: 0 (0 bytes)
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # PUSH_DATA datagrams sent: 1 (220 bytes)
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # PUSH_DATA acknowledged: 100.00%
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: ### [DOWNSTREAM] ###
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # PULL_DATA sent: 3 (100.00% acknowledged)
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # RF packets sent to concentrator: 0 (0 bytes)
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # TX errors: 0
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: ### [GPS] ###
Aug 29 20:33:56 ttn-gateway ttn-gateway[377]: # Invalid gps time reference (age: 1630262036 sec)

lostall2, 2021-09-04:
i had the same problem like @dechantw :
--> # Invalid gps time reference (age: 1630262036 sec)

The ic880a has no gps module, soI turned GPS off in:
cd opt/ttn-gateway/bin --> global_conf.json

Now it's working!

lostall2, 2021-09-10:
Now I have a new problem. After a few hours, the online status of the gateway in the TTS stack only shows "Receive Gateway Status". No more messages are received. Reset pin is set correctly. Anyone have an idea?
[image]

gonzalocasas, 2021-10-21:
Yes, now is "later in year". Go for it! ;) Updating is trivial, just update the server to the new V3 servers.

BerndGaykema, 2021-10-21:
Sorry for being stupid, but how do I do that?
I just created the gateway again, using the step-by-step instructions [here](https://github.com/ttn-zh/ic880a-gateway/wiki).
It is using the JSON file, found [here](https://github.com/ttn-zh/gateway-remote-config).

How do I update this remote json file? It is only you who can do that? It is this one (B827EBFFFE0060A9.json).
I also understand that this file is no longer up-to-date regarding what is required by V3: it requires more modern stuff, see [here](https://www.thethingsnetwork.org/forum/t/trouble-with-migration-a-rpi-ic880a-gw-to-v3/44373).

I was hoping on an updated step-by-step guidance so that it is possible for me to have the gateway activated again, but now on V3.

I would be very grateful if you could help me (and others I suppose who build the Zurich-gateway) here.
Thank you very much.

BerndGaykema, 2021-10-31:
I had my modified json file merged into the master (thanks gonzalocasas).
I started from scratch and rebuild the SD-card for my RPi, using the new global json file.

The gateway is on air again. I see the gateway data in my newly created gateway V3.

One thing was not clear, and that is the FREQ PLAN. Which to use? There are more EU options to choose from.
I took SF9 for RX2 (recommended), but there is also a SF12 for RX2. What is the difference?

Last point is "CLAIM GATEWAY". Is this recommended to do?

SchneHa, 2021-11-04:
Hello,

I tried to update to migrate my RPi/iC880a Gateway to V3 and I got the following error message:

```
Updating installer files...
error: Ihre lokalen Änderungen in den folgenden Dateien würden beim Auschecken
überschrieben werden:
	src/sensors_js/install.sh
	src/sensors_js/sensors-js.service
	src/sensors_js/sensors.js
Bitte committen oder stashen Sie Ihre Änderungen, bevor Sie Branches
wechseln.
Abbruch
```
Indeed, I did changes to the files to use some sensors and wouldn't like to loose it. What can I do?

BR,
Hans

fuma01, 2021-11-12:
hello all,

for me, the following procedure worked to migrate the ic880a gateway to V3:
1. modify your json file on https://github.com/ttn-zh/gateway-remote-config and change server to "eu1.cloud.thethings.network"
2. connect you gateway with ssh and go to directory `cd /opt/ttn-gateway/gateway-remote-config `
3. check the server in your json file with `nano B827EBXXXXXXXXXX.json`, at my case the old ttn server was written here
4. remove the whole directory "gateway-remote-config" with `cd ..` and `sudo rm -r gateway-remote-config`
5. clone the modified repository `sudo git clone https://github.com/ttn-zh/gateway-remote-config`
6. now also the "local_conf.json" at `cd /opt/ttn-gateway/bin ` should be updated with the new server
7. restart `sudo reboot`
8. gateway is connected at "eu1.cloud.thethings.network/console"