ttn-zh/ic880a-gateway #78: Service Error / Not Connected in the Things Network

symeonmattes, 2021-04-04:
Hi,

I have followed the instructions in [ic880a-gateway/](https://github.com/ttn-zh/ic880a-gateway/wiki). 

I run 

**sudo systemctl status ttn-gateway**                                                                                                
```
● ttn-gateway.service - The Things Network Gateway
   Loaded: loaded (/lib/systemd/system/ttn-gateway.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2021-04-04 23:36:30 EEST; 1s ago
  Process: 15300 ExecStart=/opt/ttn-gateway/bin/start.sh (code=exited, status=1/FAILURE)
 Main PID: 15300 (code=exited, status=1/FAILURE)

Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Ghoststream data is disabled
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Radiostream data is enabled
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Statusstream data is enabled
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Beacon is disabled
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Monitor is disabled
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Contact email configured to "symeon.mattes@gmail.com"
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Description configured to "ttn-ic880a"
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: Successfully contacted server router.eu.thethings.network
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: INFO: [main] Starting the concentrator
Apr 04 23:36:30 ttn-gateway ttn-gateway[15300]: ERROR: [main] failed to start the concentrator
```

I checked **/var/log/syslog** and I receive:
```
Apr  4 23:38:08 ttn-gateway systemd[1]: ttn-gateway.service holdoff time over, scheduling restart.
Apr  4 23:38:08 ttn-gateway systemd[1]: Stopping The Things Network Gateway...
Apr  4 23:38:08 ttn-gateway systemd[1]: Starting The Things Network Gateway...
Apr  4 23:38:08 ttn-gateway systemd[1]: Started The Things Network Gateway.
Apr  4 23:38:08 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 5: echo: write error: Device or resource busy
Apr  4 23:38:08 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 6: /sys/class/gpio/gpio25/direction: No such file or directory
Apr  4 23:38:08 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 7: /sys/class/gpio/gpio25/value: No such file or directory
Apr  4 23:38:08 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 9: /sys/class/gpio/gpio25/value: No such file or directory
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 11: /sys/class/gpio/gpio25/value: No such file or directory
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: /opt/ttn-gateway/bin/start.sh: line 13: echo: write error: Invalid argument
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: *** Poly Packet Forwarder for Lora Gateway ***
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: Version: 2.1.0
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: *** Lora concentrator HAL library version info ***
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: Version: 3.1.0; Options: native;
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: ***
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Little endian host
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: found global configuration file global_conf.json, parsing it
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: lorawan_public 1, clksrc 1
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Configuring TX LUT with 16 indexes
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: gateway MAC address is configured to AA555A0000000000
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Found 2 servers in array.
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Server 0 configured to "127.0.0.1", with port up "1680" and port down "1681"
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Server 1 configured to "iot.semtech.com", with port up "1680" and port down "1680"
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: Found 5 system calls in array.
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: System command 0: "df -m"
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: System command 1: "free -h"
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: System command 2: "uptime"
Apr  4 23:38:09 ttn-gateway ttn-gateway[15626]: INFO: System command 3: "who -a"
[... 70 lines cut ...]
In **ls -la /sys/class/gpio** I have:
```
drwxrwx---  2 root gpio    0 Apr  4 22:16 .
drwxr-xr-x 51 root root    0 Apr  4 22:23 ..
-rwxrwx---  1 root gpio 4096 Apr  4 23:44 export
lrwxrwxrwx  1 root gpio    0 Apr  4 22:16 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root gpio    0 Apr  4 22:16 gpiochip100 -> ../../devices/gpiochip2/gpio/gpiochip100
lrwxrwxrwx  1 root gpio    0 Apr  4 22:16 gpiochip128 -> ../../devices/gpiochip1/gpio/gpiochip128
-rwxrwx---  1 root gpio 4096 Apr  4 23:44 unexport
```

Furthermore in **cat /sys/kernel/debug/gpio**
```
gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-4   (                    |w1                  ) in  lo    
 gpio-25  (                    |ads7846_pendown     ) in  hi    

gpiochip2: GPIOs 100-101, brcmvirt-gpio, can sleep:
 gpio-100 (                    |?                   ) out lo    

gpiochip1: GPIOs 128-135, brcmexp-gpio, can sleep:
 gpio-135 (                    |?                   ) in  lo    
```

It seems that gpio-25 is already used.

gonzalocasas, 2021-04-04:
Perhaps you forgot to enable spi. These type of issues is not a problem with the code in this repository but always a configuration issue, either failing to enable all required options or incorrect wiring.

closed 2021-04-04 by gonzalocasas

symeonmattes, 2021-04-04:
I have already checked it the spi. it is enabled. But I am not  sure how to debug it ... is there anyway to check if the cabling is correct...or any other configuration?