ttn-zh/ic880a-gateway #84: Best place in Switzerland to find the iC880a board ?

vincegre, 2021-09-16:
Hi

Planning to build a gw based on your design but I'm unable to find the board in Switzerland ! Any advice to buy it at best price ? or a shop in Europe ? I would need the board and  a good antenna to go with ;)

Thanks

Vincèn

gonzalocasas, 2021-09-27:
It is now available here: https://www.tindie.com/products/iotdevices/imst-ic880a-lorawan-backplane/

closed 2021-09-27 by gonzalocasas

vincegre, 2021-09-27:
> It is now available here: https://www.tindie.com/products/iotdevices/imst-ic880a-lorawan-backplane/

Thanks, will try to get an IC880a board first as it looks nearly impossible to find some stock of it :(

gonzalocasas, 2021-09-27:
The RAK wireless pi hat for lorawan are compatible, so you might want to try those.