ttn-zh/ic880a-gateway #91: Error: ERROR: [main] failed to start the concentrator

heydemar, 2022-08-31:
Hello,
After my SD card crashed, I had to set up my LORAWAN Gateway again and followed the instructions as before, but unfortunately I now get the following error message when I look at the status with the following command (SPI is on):

sudo journalctl -u ttn-gateway -f

-------------

Aug 31 10:58:27 LORAWAN ttn-gateway[2280]: INFO: [main] Starting the concentrator
Aug 31 10:58:27 LORAWAN ttn-gateway[2280]: ERROR: [main] failed to start the concentrator
Aug 31 10:58:27 LORAWAN systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
Aug 31 10:58:27 LORAWAN systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
Aug 31 10:58:32 LORAWAN systemd[1]: ttn-gateway.service: Scheduled restart job, restart counter is at 24.
Aug 31 10:58:32 LORAWAN systemd[1]: Stopped The Things Network Gateway.
Aug 31 10:58:32 LORAWAN systemd[1]: Started The Things Network Gateway.

marcobrianza, 2022-11-28:
Hi, I have the same issue. @heydemar did you find a solution?

marcobrianza, 2022-11-28:
I fixed the issue enabling SPI