ttn-zh/ic880a-gateway #93: Errors while launching start.sh to restart the gateway

green15878, 2024-06-03:
$ sudo /opt/ttn-gateway/bin/start.sh

/opt/ttn-gateway/bin/start.sh: line 5: echo: write error: Invalid argument
/opt/ttn-gateway/bin/start.sh: line 6: /sys/class/gpio/gpio25/direction: No such file or directory
/opt/ttn-gateway/bin/start.sh: line 7: /sys/class/gpio/gpio25/value: No such file or directory
/opt/ttn-gateway/bin/start.sh: line 9: /sys/class/gpio/gpio25/value: No such file or directory
/opt/ttn-gateway/bin/start.sh: line 11: /sys/class/gpio/gpio25/value: No such file or directory
/opt/ttn-gateway/bin/start.sh: line 13: echo: write error: Invalid argument
/opt/ttn-gateway/bin/start.sh: line 53: ./poly_pkt_fwd: No such file or directory

gonzalocasas, 2024-06-03:
This is most likely a misconfiguration

green15878, 2024-06-03:
i followed all the steps , any suggested solutions ?

dusk0r, 2025-01-19:
` /sys/class/gpio` is obsolete and was removed. Someone should update the script.

gonzalocasas, 2025-01-20:
Would you have time to send a pull request with the fix?

dusk0r, 2025-01-21:
https://github.com/ttn-zh/ic880a-gateway/pull/94

Works on my TTN gateway with current Raspberry PI OS Lite (Kernel 6.6.62).